pybricks/support #1009: [Bug] [pybricksdev] Second connected hub program download is significantly slower

Novakasa, 2023-03-25:
**Describe the bug**

It seems that maintaining two BLE connections to pybricks hubs at the same time slows down communication with the first connected device significantly. Specifically, I am seeing a reduction in program download speed in pybricksdev.

The reduction happens only for the first connected device, only while two hubs are connected. Disconnecting the second device makes the download for the first device fast again.

First described in https://github.com/pybricks/support/issues/971#issuecomment-1480134346, @dlech referenced https://github.com/hbldh/bleak/issues/970

**To reproduce**

Using pybricksdev, hub program:

```python

from pybricks.tools import wait

data = """
Lorem ipsum dolor sit amet, consetetur sadipscing elitr, sed diam nonumy eirmod tempor invidunt ut labore et dolore magna aliquyam erat, sed diam voluptua. At vero eos et accusam et justo duo dolores et ea rebum. Stet clita kasd gubergren, no sea takimata sanctus est Lorem ipsum dolor sit amet. Lorem ipsum dolor sit amet, consetetur sadipscing elitr, sed diam nonumy eirmod tempor invidunt ut labore et dolore magna aliquyam erat, sed diam voluptua. At vero eos et accusam et justo duo dolores et ea rebum. Stet clita kasd gubergren, no sea takimata sanctus est Lorem ipsum dolor sit amet. Lorem ipsum dolor sit amet, consetetur sadipscing elitr, sed diam nonumy eirmod tempor invidunt ut labore et dolore magna aliquyam erat, sed diam voluptua. At vero eos et accusam et justo duo dolores et ea rebum. Stet clita kasd gubergren, no sea takimata sanctus est Lorem ipsum dolor sit amet.   

Duis autem vel eum iriure dolor in hendrerit in vulputate velit esse molestie consequat, vel illum dolore eu feugiat nulla facilisis at vero eros et accumsan et iusto odio dignissim qui blandit praesent luptatum zzril delenit augue duis dolore te feugait nulla facilisi. Lorem ipsum dolor sit amet, consectetuer adipiscing elit, sed diam nonummy nibh euismod tincidunt ut laoreet dolore magna aliquam erat volutpat.   

Ut wisi enim ad minim veniam, quis nostrud exerci tation ullamcorper suscipit lobortis nisl ut aliquip ex ea commodo consequat. Duis autem vel eum iriure dolor in hendrerit in vulputate velit esse molestie consequat, vel illum dolore eu feugiat nulla facilisis at vero eros et accumsan et iusto odio dignissim qui blandit praesent luptatum zzril delenit augue duis dolore te feugait nulla facilisi.   

Nam liber tempor cum soluta nobis eleifend option congue nihil imperdiet doming id quod mazim placerat facer possim assum. Lorem ipsum dolor sit amet, consectetuer adipiscing elit, sed diam nonummy nibh euismod tincidunt ut laoreet dolore magna aliquam erat volutpat. Ut wisi enim ad minim veniam, quis nostrud exerci tation ullamcorper suscipit lobortis nisl ut aliquip ex ea commodo consequat.   

Duis autem vel eum iriure dolor in hendrerit in vulputate velit esse molestie consequat, vel illum dolore eu feugiat nulla facilisis.   

At vero eos et accusam et justo duo dolores et ea rebum. Stet clita kasd gubergren, no sea takimata sanctus est Lorem ipsum dolor sit amet. Lorem ipsum dolor sit amet, consetetur"""

wait(1000)
```

PC program:

```python

import asyncio
from pybricksdev.ble import find_device
from pybricksdev.connections.pybricks import PybricksHub

async def main():

    dev1 = await find_device()
    hub1 = PybricksHub()
    await hub1.connect(dev1)
    print("hub1 connected!")

    print("hub1 downloading...")
    await hub1.run("brickrail-gui/ble-server/hub_programs/test_noop_large.py")
    await asyncio.sleep(1)

    dev2 = await find_device()
    hub2 = PybricksHub()
    await hub2.connect(dev2)
    print("hub2 connected!")

    print("hub1 downloading...")
    await hub1.run("brickrail-gui/ble-server/hub_programs/test_noop_large.py")

    print("hub2 downloading...")
    await hub2.run("brickrail-gui/ble-server/hub_programs/test_noop_large.py")

    await hub2.disconnect()
    print("hub2 disconnected!")

    print("hub1 downloading...")
    await hub1.run("brickrail-gui/ble-server/hub_programs/test_noop_large.py")

    await hub1.disconnect()
    print("hub1 disconnected!")

asyncio.run(main())
```

Output:

```
hub1 connected!
hub1 downloading...
100%|█████████████████████████████████████████████████████████████████████████████████████████████████| 2.58k/2.58k [00:02<00:00, 1.00kB/s]
hub2 connected!
hub1 downloading...
100%|███████████████████████████████████████████████████████████████████████████████████████████████████| 2.58k/2.58k [00:07<00:00, 346B/s]
hub2 downloading...
100%|█████████████████████████████████████████████████████████████████████████████████████████████████| 2.58k/2.58k [00:02<00:00, 1.13kB/s]
hub2 disconnected!
hub1 downloading...
100%|█████████████████████████████████████████████████████████████████████████████████████████████████| 2.58k/2.58k [00:02<00:00, 1.10kB/s]
hub1 disconnected!
```

You can see in the reported download speed that all downloads are reported as > 1kB/s except for the download on hub1, only when both hubs are connected.

**Expected behavior**
I am expecting that all hubs download at a similar rate, independent of which hub I connected first.

laurensvalk, 2023-03-25:
I'm quite sure I've seen this with Pybricks Code too. I'd have to check if it is the the first device in this case too.

dlech, 2023-03-28:
This may also be a hint to explain some of the issues we have had reported with macOS recently when other devices are connected as in #938.

dlech, 2023-03-28:
So, in conclusion, I think this is just expected behavior of BLE in general.

closed 2023-03-28 by dlech