pybricks/support #1015: [Bug] Running Motor.run_until_stalled

oemtas, 2023-03-28:
**Describe the bug**
I get `The requested operation is not valid in the current state` when running motor.run_until_stalled(-999, Stop.BRAKE, 5)

**Screenshots**
[image]
[image]

oemtas, 2023-03-28:
To your information: The Motor is not used in a Drivebase

laurensvalk, 2023-03-28:
Thanks for reporting. Can you share your full program please?

Or a small program where this happens?

oemtas, 2023-03-28:
```
#!/usr/bin/env pybricks-micropython
from pybricks.ev3devices import Motor, ColorSensor
from pybricks.hubs import EV3Brick
from pybricks.parameters import Port, Color, Stop
from pybricks.robotics import DriveBase

ev3 = EV3Brick()
motor = Motor(Port.A)
wheel_right = Motor(Port.B)
wheel_left = Motor(Port.C)

drivebase = DriveBase(left_motor=wheel_left, right_motor=wheel_right, wheel_diameter=50, axle_track=210)
drivebase.settings(straight_speed=999, turn_rate=999)

def run():
    drivebase.straight(50)
    //Happens here
    #motor.run_until_stalled(-999, Stop.BRAKE, 5)

run()
```

laurensvalk, 2023-03-28:
I think you need to use a higher value for the `duty_limit` than `5`. Perhaps try something like `30`.

oemtas, 2023-03-28:
Okay, worked. Thank you

laurensvalk, 2023-03-28:
Glad to help!

closed 2023-03-28 by laurensvalk