pybricks/support #1016: Starting program failed: Failed to execute child process "/home/robot/ev3python/BarcodeTester\BarcodeReader" (No such file or directory)

green4ja, 2023-03-28:
**Describe the bug**
After changing literally nothing about how my folders and directories are pathed. I created a new folder within my ev3python parent folder used for holding my BarcodeReader code in the BarcodeTester folder. Every time I try to run the file in run&debug mode or just on the actual ev3 mindstorm, the terminal outputs this:

`Starting: brickrun -r --directory="/home/robot/ev3python" "/home/robot/ev3python/BarcodeTester\BarcodeReader"
Starting program failed: Failed to execute child process "/home/robot/ev3python/BarcodeTester\BarcodeReader" (No such file or directory)
Exited with error code 1.`

**Context**
I was just trying to test out the barcode reading code with my color sensor.

**Screenshots**
[image]
[image]
[image]

green4ja, 2023-03-28:
**More Info**
Quick bit of info, I don't have the slightest clue about how directories and folders and file paths work in vscode, and 99% of the time they are the source of my errors and lead to me spending endless time trying to figure out a problem I have no clue how to solve.

**Directory Info**
The directory for the folder with the file I'm trying to run is:
`C:\Users\jag10\Desktop\Stuff for Python Robot\ev3python\BarcodeTester`
(Also, please let me know if this is an incorrect way of doing things, but everytime when I go to work on my code for these robot files in vscode, I click Open Folder, then chose my ev3python folder that has all my subfolders with the code files in them.)

**Screenshot**
This is the screenshot of the inside of ev3python folder:
[image]

dlech, 2023-03-29:
Can you share your `.vscode/launch.json` file? I suspect you need to replace `\\` with `/` in the `"program"` listed there.

closed 2023-03-29 by dlech

green4ja, 2023-03-29:
Wait why was this closed? I still don't know what to change?

dlech, 2023-03-29:
It was closed automatically because I just pushed a fix and I didn't have time to comment yet to explain. 😄 

Please try updating the `ev3dev-browser` to v1.2.1.

The `${relativeFile}` variable includes the Windows path separator which doesn't work on EV3 since it is running Linux. So I added a fix to automatically convert the path separators.

green4ja, 2023-03-29:
I checked my extensions and ev3dev-browser is on v1.2.1. Also I don't really know how to incorporate your fix into my code.

dlech, 2023-03-29:
Did you try running the program again now that the extension has been updated? You should not have to change anything in your code.

green4ja, 2023-03-29:
I'm currently trying to turn on my robot to run the program directly to the robot but of course for the first time my robot has been stuck on a timed out message the last 3 boots.

green4ja, 2023-03-29:
It works now! Thank you so much!

dlech, 2023-03-29:
Great! Thanks for confirming!