pybricks/support #1018: [Doc] Add tutorial for Visual Studio Code usage with pybricksdev and pybricks-api

laurensvalk, 2023-03-29:
**Describe the issue**

We get requests for Visual Studio Code usage every once in a while.

It would be convenient to have this on our site so we can link to it.

We could also explain that Pybricks Code is the official way to do it, but pybricksdev+pybricks-api can be used for download and run and autocomplete.

**Suggested improvement**

[image]

laurensvalk, 2023-03-29:
To get us going, I have added a placeholder for the tutorial, including a relevant icon and (still empty) `.md` file in https://github.com/pybricks/pybricks-projects/tree/master/tutorials/dev/tools/vscode.

@dlech, is this perhaps something you'd want to do? I think you have explained how to do this in several open discussions; maybe we could copy those explanations there?

dlech, 2023-03-30:
done in https://github.com/pybricks/pybricks-projects/commit/99f7b8b8534c68e13cb3041e136f368c8fb1f242

closed 2023-03-30 by dlech