pybricks/support #1022: [Feature] Review IMU sample rate

laurensvalk, 2023-04-01:
This task is split from https://github.com/pybricks/pybricks-micropython/pull/156 so we can look at it separately.

----

> https://github.com/pybricks/pybricks-micropython/blob/0a0332f5a9cb8b33b9d8f67f85badf51a408331b/lib/pbio/drv/imu/imu_lsm6ds3tr_c_stm32.c#L203
>
> We probably need to verify what the actual sampling rate is, or measure it as we go. Similarly, we'll want to check if we are occasionally missing a few samples or not.
>
> We'll also need to check this on Technic Hub, where the I2C setup is slightly different, in case there are any difference.
>
> Once that is settled, we may need to experimentally verify `lsm6ds3tr_c_from_fs2000dps_to_mdps` and adjust as needed. Possibly per axis, see also below.

----

> There is currently no adjustment for misalignment of the chip within the hub. This is possible using a calibration routine (see https://github.com/pybricks/support/issues/943), but the benefit is fairly small for everyday users.
>
> However, it may be useful to do this locally when we test for sample rates as above, to ensure we register the full Z measurement when tweaking `lsm6ds3tr_c_from_fs2000dps_to_mdps`.

---

> In `pbio`, there is currently not a global `pbio_imu` instance that gets passed around, since there is only one `imu` instance. It may still be useful to make the code easier to follow. 

---

_Originally posted by @laurensvalk in https://github.com/pybricks/pybricks-micropython/issues/156#issuecomment-1488174256_

laurensvalk, 2023-04-01:
> Hmm... if [init] fails, won't calling other functions later cause a crash due to uninitialized values?

_Originally posted by @dlech in https://github.com/pybricks/pybricks-micropython/pull/156#discussion_r1152234070_

laurensvalk, 2023-04-01:
> Should raise `NotImplementedError` instead?

_Originally posted by @dlech in https://github.com/pybricks/pybricks-micropython/pull/156#discussion_r1152281152_

Intuitive drivebase integration still needs a bit of work. The `COUNTERCLOCKWISE` direction should probably also be done at the `pbio` level, not the module as we do know.

laurensvalk, 2023-04-01:
> Since this depends on the IMU, it seems like it should be a separate IMU module rather part of the orientation module (separating the "pure" functions from the hardware dependent functions).

_Originally posted by @dlech in https://github.com/pybricks/pybricks-micropython/pull/156#discussion_r1152212982_

May need to revisit this. Instead we can probably move the pure functions in `orientation` to `geometry`.

laurensvalk, 2023-04-01:
> Probably can do without the `union` here (we can cast to `float *` if we really need to but better if we just use the struct everywhere).
>
> Or if we really must keep the union, the outer `struct` is redundant and we can just say `typedef union {`.

_Originally posted by @dlech in https://github.com/pybricks/pybricks-micropython/pull/156#discussion_r1152138645_

laurensvalk, 2023-04-01:
This also still needs testing on Technic Hub and Essential Hub

laurensvalk, 2023-04-01:
On Technic Hub, it never becomes stationary with the default settings. So as per OP, we need to dig into the sensor configuration a bit more.

laurensvalk, 2023-04-01:
> We probably need to verify what the actual sampling rate is, or measure it as we go. Similarly, we'll want to check if we are occasionally missing a few samples or not.

I wonder if the scale varies slightly per hub. One one hub, the heading seems to be off by a few degrees after one full rotation.

dlech, 2023-04-01:
- [ ] Rename `orientation.h` to `imu.h` 
- [ ] Drop `#include <pbdrv/imu.h>` and fix copyright date  in above
- [ ] Rename `pbio_orientation_set_base_orientation()` to `pbio_imu_set_base_orientation()`
- [ ] Mark `pbio_imu_init()` as internal (doxygen `@condition INTERNAL``) or move to separate header file.

laurensvalk, 2023-04-02:
Yeah, those are already done and pushed, just not merged to master yet.

laurensvalk, 2023-04-04:
> It would be easy enough to compensate by measuring the frequency at runtime, but is there an explanation for this?

Done via https://github.com/pybricks/pybricks-micropython/commit/1dbc334c904657310413d462bbdc6c71b713d424

dlech, 2023-04-04:
> but is there an explanation for this?

Since the IMU doesn't have an external crystal oscillator, there is never going to be an accurate (as noted) or stable (e.g. could change with temperature) clock on the IMU chip.

laurensvalk, 2023-04-07:
- [ ] Update settings setter to match API in https://github.com/pybricks/pybricks-api/pull/137/commits/28ecc060b72091472e9803ec8e3883c6d05e8077

laurensvalk, 2023-04-07:
The most critical parts here are now done and the implementation now matches the API proposal in https://github.com/pybricks/pybricks-api/pull/137

laurensvalk, 2024-10-17:
This is done.

closed 2024-10-17 by laurensvalk