pybricks/support #1029: Unexpected turning problems, possibly a gyro sensor issue.

green4ja, 2023-04-07:
**Question**
Why is my robot's second turn faster than the first, and why does the robot start acting weird after that turn?

**Context**
I'm trying to get my robot to follow the path of a square infinitely as a way of testing the precision of my motors and gyro sensor. Everything works fine for the first straight and turn, but the second turn for whatever reason is at a faster speed than the first and much less precise, and after this point, it starts acting quite unexpectedly and erratic.

**The Code**

[image]
[image]

**Conclusion**
I'd really appreciate some feedback on this issue, as I'm trying to learn the ropes of ev3 micropython, thanks in advance!

laurensvalk, 2023-04-07:
To better understand what the sensor is doing, try creating a separate program that just prints out the values. Alternatively, watch the values in the EV3 on-screen menu.

The EV3 sensor tends to have a lot of drift issues, where the value moves away even as it sits still. Could this explain what is going on?

In some cases it helps unplugging the cable and plugging it back in, while ensuring the sensor is still as you do so.

green4ja, 2023-04-07:
Thank you for providing some possible solutions, unfortunately, I'm not able to test anything for a little while but I would still like to clarify some things you said. In the wait(10) function, is 10 in milliseconds or seconds? Also, is it usually as simple as just not moving your ev3 brick during startup that fixes gyro drift?

laurensvalk, 2023-04-07:
> In the wait(10) function, is 10 in milliseconds or seconds? 

milliseconds

> Also, is it usually as simple as just not moving your ev3 brick during startup that fixes gyro drift?

The sensor needs to be kept still. It doesn't matter if the EV3 brick is moving.

green4ja, 2023-04-08:
So, I got to changing these things in my code and being aware of keeping the sensor still when booting up. I also check my gyro sensor values on the ev3 brick before and after each test to make sure there isn't gyro drift, and calibrate it if needed. Even with doing all this, for some reason, the second gyro_turn() is always done at a faster speed and it seems to blend right into the next gyro_straight as well. I have a feeling it is somehow a problem with the code.

closed 2025-07-13 by dlech