pybricks/support #1030: non-LEGO dc-motor on Technic Hub

Werner-G, 2023-04-09:
Hi all, I am not able to get a non-LEGO DC-motor (Geekservo) work on the technic hub. The hardware:
1) motor between pin1 and pin3 (M1 and GND) or
2) motor between pin1 and pin3, and pin5 (ID1) via 2k2 to GND or
3) motor between pin1 and pin3, and pin5 (ID1) via 2k2 to GND, and pin6 (ID2) to GND

the demo code is:
```
from pybricks.hubs import TechnicHub
from pybricks.pupdevices import DCMotor, Motor, Remote
from pybricks.parameters import Button, Color, Port, Stop
from pybricks.iodevices import PUPDevice
from uerrno import ENODEV

hub = TechnicHub()

left_motor = Motor(Port.A)
right_motor = Motor(Port.D)
red_motor = DCMotor(Port.C)

red_motor.dc(60)
while True:
    wait(3000)
```
The error message is: A sensor or motor is not connected to the specified port.
Any idea what is wrong?
and, btw: any example if any non-LEGO-sensor can be attached to the hub, using I²C?

laurensvalk, 2023-04-10:
This is how to wire a DC motor:

- Pin 1 + Pin 2 go to the DC motor.
- Pin 6 to GND = PIN 3
- Pin 5 to VCC = P4

This will make it appear like a "train" motor that you can use with the `DCMotor` class.

If you can try it out before you solder it together, you can flip M1/M2 if needed so that the default direction matches that of the `DCMotor` class: A positive duty cycle value should make it turn clockwise when looking at the shaft.

Werner-G, 2023-04-10:
Thanks you. Still, the motor doesn't move. Is the demo code (above) right?

laurensvalk, 2023-04-10:
The code looks good to me. Does it still work with a normal motor?

Does it still give the `ENODEV` exception with your custom motor? If it gets beyond that, can you measure a voltage across M1/M2?

Werner-G, 2023-04-10:
no error message. The motor is OK, runs at Pins 3 and 4. In the moment, I cannot use another motor or measure anything. Will do it in two days.

laurensvalk, 2023-04-10:
Glad to hear it worked!

closed 2023-04-10 by laurensvalk