pybricks/support #1034: [Feature] add `pybricks-micropython` binary to the release assets

kai-morich, 2023-04-16:
I succesfully installed different `pybricks-micropython` on EV3 with the help of https://github.com/pybricks/support/issues/648.

Finding the file for the master branch works as described in #648, but getting that file for release versions is a bit tricky, e.g. works for recent [v3.2.3](https://github.com/pybricks/pybricks-micropython/actions/runs/4207803050) but not for older releases like [v3.1.0](https://github.com/pybricks/pybricks-micropython/actions/runs/1589591238) because Github keeps build artifacts only for 90 days.

Would be nice if you could add the `pybricks-micropython` binary to the release assets.

dlech, 2023-05-26:
To make easier to install, I've published a new debian package for the latest beta release (v3.3.0b5).

To get it, run the following on an EV3 that is connected to the Internet:

    sudo apt update
    sudo apt install pybricks-micropython3

If you want to use this version in your programs, change the shebang to `#!/usr/bin/env pybricks-micropython3` (the only difference is the `3` at the end).

Note: there will be errors reported in `apt update` due to debian stretch being out of support but they can be ignored. If it really bother you and you want to fix it or need other packages, you can fix it as follows:

```
wget https://raw.githubusercontent.com/ev3dev/docker-library/master/ev3dev-stretch/layers/debian/etc/apt/sources.list
sudo mv sources.list /etc/apt/
sudo apt update
```

kai-morich, 2023-05-28:
That solution is really cool. All can be done now in VS Code. No more downloading, copy to ev3, forgetting x flag, ...
I don't mind the warning from `apt update`.
Biggest effort was configuring internet access by precisely following these steps: https://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-usb/

Would be good if you document this somewhere, e.g. link [here](https://github.com/pybricks/pybricks-micropython#supported-programmable-bricks--hubs) as different variants for EV3. 
BTW the link there to _latest ev3dev image_ is broken. I simply used the version from LEGO and above _sudo apt install ..._

Tear4Pixelation, 2025-04-27:
@dlech sorry but is the `pybricks-micropython3` package still available? For me it just says
`E: Unable to locate package pybricks-micropython3`. I could really use v3 as it has some features I need.
As far as I am concerned, I have the newest sd card image (from https://www.ev3dev.org/docs/getting-started/)

dlech, 2025-04-27:
You have to have the EV3 connected to the Internet and do `sudo apt update` first. Also, beware that motor control is not working well in Pybricks v3.x on ev3dev, which is why we dropped support for it.

Tear4Pixelation, 2025-04-27:
@dlech Well I did run `sudo apt update` and it didn't say 404 or some other connection-based issue, so I think the internet works. The thing is that the `DriveBase.arc()` and a few others are needed for my project (currently I am relying on my own motor control system, but I have no way to add acceleration there, as it is pure python, and I cant get my robot to drive straight *frustration*)

dlech, 2025-04-27:
It's not quite ready for testing yet, but we hope to have an alternate solution soon. https://pybricks.com/project/saving-lego-mindstorms/

Tear4Pixelation, 2025-04-28:
@dlech  that’s so cool! But there’s still one question in my mind then… it seems that you’re trying to support the web interface of Pybricks. Is very good for people that need a plug and play experience, but I really like my visual studio code with the ev3dev browser extension. Will something like that still be supported or do I have to use the web interface (also fine)?
Best of luck with that!

dlech, 2025-04-28:
We will make it work with the `pybricksdev` command line tool as well.

dlech, 2025-07-13:
Closing this since we're dropping support for ev3dev in favor of a new EV3 solution.