pybricks/support #1039: No Intellisense in VSCode on Linux

Tear4Pixelation, 2023-04-21:
I really need it :(

It could be a linux problen though

dlech, 2023-04-21:
Have you seen https://pybricks.com/projects/tutorials/dev/tools/vscode/?

Tear4Pixelation, 2023-04-21:
> Have you seen https://pybricks.com/projects/tutorials/dev/tools/vscode/?
WOW Thanks for the amazingly quick answer.
What should I change? I've set the language to pylance, but that did not change anything

dlech, 2023-04-21:
There are more steps like creating a virtual environment, installing the `pybricks` package, editing the `.vscode/settings.json` (only if you created the project using the EV3 extension). Did you do all of those steps too?

Tear4Pixelation, 2023-04-21:
> There are more steps like creating a virtual environment, installing the `pybricks` package, editing the `.vscode/settings.json` (only if you created the project using the EV3 extension). Did you do all of those steps too?

Sorry, I am working on a EV3 Project right now, so I might not be too responsive.
I got the Pybricks add-on, but the virtual environment, should it look like this?
[image]

Tear4Pixelation, 2023-04-21:
I will most probably not go online today again, so don't expect replies.

dlech, 2023-04-21:
> should it look like this?

No, it looks like you have not created a virtual environment yet. The selected Python interpreter is still the system default and there is no `.venv` folder.

Tear4Pixelation, 2023-04-21:
> > should it look like this?
> 
> No, it looks like you have not created a virtual environment yet. The selected Python interpreter is still the system default and there is no `.venv` folder.

(I went on against all odds)
What do I have to do? (Sorry I'm quite the noob)

dlech, 2023-04-21:
The link I gave earlier has detailed instructions on how to do this.

Tear4Pixelation, 2023-04-21:
I get this
Experiment 'pythonPromptNewFormatterExt' is active
Experiment 'pythonPromptNewToolsExt' is active
LSP Notebooks experiment is disabled -- not using Pylance
LSP Notebooks interactive window support is disabled -- not in LSP Notebooks experiment
Python interpreter path: /bin/python3
Editor support is inactive since language server is set to None.
> conda info --json
> /bin/python3 ~/.vscode/extensions/ms-python.python-2022.20.2/pythonFiles/get_output_via_markers.py ~/.vscode/extensions/ms-python.python-2022.20.2/pythonFiles/interpreterInfo.py
> /bin/python3.9 ~/.vscode/extensions/ms-python.python-2022.20.2/pythonFiles/get_output_via_markers.py ~/.vscode/extensions/ms-python.python-2022.20.2/pythonFiles/interpreterInfo.py
> /usr/bin/python3 ~/.vscode/extensions/ms-python.python-2022.20.2/pythonFiles/get_output_via_markers.py ~/.vscode/extensions/ms-python.python-2022.20.2/pythonFiles/interpreterInfo.py
> /usr/bin/python3.9 ~/.vscode/extensions/ms-python.python-2022.20.2/pythonFiles/get_output_via_markers.py ~/.vscode/extensions/ms-python.python-2022.20.2/pythonFiles/interpreterInfo.py
Running Env creation script:  [
  '/bin/python3',
  '/home/jan/.vscode/extensions/ms-python.python-2022.20.2/pythonFiles/create_venv.py',
  '--git-ignore',
  '--install'
]
Running: /bin/python3 -m venv .venv

The virtual environment was not created successfully because ensurepip is not
available.  On Debian/Ubuntu systems, you need to install the python3-venv
package using the following command.

    apt install python3.10-venv

You may need to use sudo with that command.  After installing the python3-venv
package, recreate your virtual environment.

Failing command: ['/home/jan/Documents/python/mindstrm/Ev3/.venv/bin/python3', '-Im', 'ensurepip', '--upgrade', '--default-pip']

Traceback (most recent call last):
  File "/home/jan/.vscode/extensions/ms-python.python-2022.20.2/pythonFiles/create_venv.py", line 60, in run_process

    subprocess.run(args, cwd=os.getcwd(), check=True)
  File "/usr/lib/python3.10/subprocess.py", line 524, in run

    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['/bin/python3', '-m', 'venv', '.venv']' returned non-zero exit status 1.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/jan/.vscode/extensions/ms-python.python-2022.20.2/pythonFiles/create_venv.py", line 136, in <module>

    main(sys.argv[1:])
  File "/home/jan/.vscode/extensions/ms-python.python-2022.20.2/pythonFiles/create_venv.py", line 122, in main
    run_process(
  File "/home/jan/.vscode/extensions/ms-python.python-2022.20.2/pythonFiles/create_venv.py", line 62, in run_process
    
raise VenvError(error_message)
__main__.VenvError: CREATE_VENV.VENV_FAILED_CREATION

[ERROR 2023-3-21 20:50:51.412]: Error while running venv creation script:  CREATE_VENV.VENV_FAILED_CREATION
[ERROR 2023-3-21 20:50:51.412]: CREATE_VENV.VENV_FAILED_CREATION
> ./.venv/bin/python ~/.vscode/extensions/ms-python.python-2022.20.2/pythonFiles/get_output_via_markers.py ~/.vscode/extensions/ms-python.python-2022.20.2/pythonFiles/interpreterInfo.py
Editor support is inactive since language server is set to None.
Python interpreter path: ./.venv/bin/python
Running Env creation script:  [
  '/bin/python3',
  '/home/jan/.vscode/extensions/ms-python.python-2022.20.2/pythonFiles/create_venv.py',
  '--git-ignore',
  '--install'
]
EXISTING_VENV:/home/jan/Documents/python/mindstrm/Ev3/.venv/bin/python

Running: /home/jan/Documents/python/mindstrm/Ev3/.venv/bin/python -m pip install --upgrade pip

/home/jan/Documents/python/mindstrm/Ev3/.venv/bin/python: No module named pip

Traceback (most recent call last):
  File "/home/jan/.vscode/extensions/ms-python.python-2022.20.2/pythonFiles/create_venv.py", line 60, in run_process

    subprocess.run(args, cwd=os.getcwd(), check=True)
  File "/usr/lib/python3.10/subprocess.py", line 524, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['/home/jan/Documents/python/mindstrm/Ev3/.venv/bin/python', '-m', 'pip', 'install', '--upgrade', 'pip']' returned non-zero exit status 1.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/jan/.vscode/extensions/ms-python.python-2022.20.2/pythonFiles/create_venv.py", line 136, in <module>
    main(sys.argv[1:])
  File "/home/jan/.vscode/extensions/ms-python.python-2022.20.2/pythonFiles/create_venv.py", line 132, in main
    install_packages(venv_path)
  File "/home/jan/.vscode/extensions/ms-python.python-2022.20.2/pythonFiles/create_venv.py", line 78, in install_packages
    run_process(
  File "/home/jan/.vscode/extensions/ms-python.python-2022.20.2/pythonFiles/create_venv.py", line 62, in run_process
    raise VenvError(error_message)
__main__.VenvError: 
CREATE_VENV.PIP_UPGRADE_FAILED

[ERROR 2023-3-21 20:51:16.476]: Error while running venv creation script:  CREATE_VENV.PIP_UPGRADE_FAILED
[ERROR 2023-3-21 20:51:16.476]: CREATE_VENV.PIP_UPGRADE_FAILED

Tear4Pixelation, 2023-04-21:
But I have pip installed

dlech, 2023-04-21:
The error message explains how to fix it:

    apt install python3.10-venv

dlech, 2023-04-21:
Probably should be `sudo apt install python3-venv` though.

Tear4Pixelation, 2023-04-21:
> 

Sorry, I did not read it. Sorry If I was annoying :(

Tear4Pixelation, 2023-04-21:
I'm going to ask my father because I don't have the superuser rights, I will report any still existing errors tomorrow

Tear4Pixelation, 2023-04-21:
Thank you for the great, fast and informative help so far!

Tear4Pixelation, 2023-04-24:
That did actually **not** help for me, but for the people wondering how to fix: you close the workspace created with the Ev3 plugin and open the folder as a normal workspace using **File > Open Folder**, and it worked fine for me