pybricks/support #1046: [Bug] Pybricks Code scripts in a bad state

laurensvalk, 2023-04-25:
**Describe the bug**
Occasionally, the script gets in a bad state. From what I can tell, several scripts got merged.

[image]

**To reproduce**
Haven't found a consistent approach. If I had to guess, this happens only when closing quite a few tabs fairly quickly. (Mouse on first one, repeatedly click middle button.)

laurensvalk, 2023-04-25:
Found it! Closing by middle-click has the side effect of paste-last-selection which is also done with middle click.

See demo below. So, depending on what was selected, this can feel like "multiple scripts merging" even if something else is happening.

For what it's worth, separate from this issue, my vote would be to disable paste-last-selection-by-middle-click altogether. I always disable in in VS Code too. It happens way too often accidentally while scrolling/moving.

https://user-images.githubusercontent.com/12326241/234357855-aba7ef45-314d-4649-8191-d487d579b739.mp4

dlech, 2023-04-25:
Nice find. I didn't even know that was a feature.

dlech, 2023-04-25:
Where/how do you disable this in VS Code? Web searches tell me this is a Linux feature rather than a VS Code feature.

laurensvalk, 2023-04-25:
I think you're right. I must be confusing it with some other VSCode feature I have disabled.

laurensvalk, 2023-04-25:
FWIW, I found that other setting, and it is indeed somewhat related:

I set `"editor.emptySelectionClipboard": false,` so that `CTRL+C` with no selection doesn't copy the line.

Sometimes I accidentally press `CTRL+C` when I mean to paste. But by default the clipboard content is now overridden so you have to start over. I suppose my Pybricks Code scripts aren't that big so it hasn't been as much of a nuissance.

dlech, 2023-04-25:
It looks like `editor.selectionClipboard` is the option. https://github.com/microsoft/vscode/issues/14610#issuecomment-256778712

Although that doesn't actually seem to fix the problem.

closed 2023-04-27 by dlech