pybricks/support #1056: how to wait for `stop()`?

kai-morich, 2023-04-29:
with this example program

```python
from pybricks.hubs import PrimeHub
from pybricks.pupdevices import Motor
from pybricks.parameters import Direction, Port, Color
from pybricks.robotics import DriveBase
from pybricks.tools import wait

hub = PrimeHub()
left_motor = Motor(Port.E, positive_direction=Direction.COUNTERCLOCKWISE)
right_motor = Motor(Port.F)
robot = DriveBase(left_motor, right_motor, 55, 80)

def set_speed(speed):
    robot.settings(speed, speed*8, speed//2, speed*5 )

# drive forward with line / wall follower controlled turn_rate
hub.light.on(Color.YELLOW)
robot.reset()
while robot.distance() < 100:
    robot.drive(300, 0)
    wait(10)

# stop
hub.light.on(Color.GREEN)
#robot.stop() # <---
#wait(300)    # <---

# drive back
hub.light.on(Color.RED)
set_speed(100)
robot.straight(-100)
```

I get different stop behavior depending on the commented out `< ---` lines:
- w/o stop(): soft decelerate with new target speed, then drive in reverse direction
- with stop(): soft decelerate, then drive in reverse direction
- with stop() + wait(100): partial hard stop, then continue forward with soft decelerate, then drive in reverse direction
- with stop() + wait(300): hard stop immediate, then drive in reverse direction

How to wait for `stop()` to be finished? 
Is there something better than
```python
robot.stop()
while(robot.state()[1]>10):
    wait(10)
```

<hr>

BTW:
Above example works on spike _and_ ev3 pybricks-micropython nightly build from #1035.  
With older ev3 pybricks-micropython versions the behavior was suboptimal and `wait()` solution was found by my WRO team by chance.

- v3.3b3
  - not testable due to initialization issues
- v3.2.3
  - w/o stop(): soft decelerate, then drive in reverse direction
  - with stop(): __fast accelerate forward for some cm__, then soft decelerate, then drive in reverse direction
  - with stop() + wait(100): partial hard stop, __fast accelerate forward for some cm__, then continue forward with soft decelerate, then drive in reverse direction
  - with stop() + wait(300): hard stop immediate, then drive in reverse direction
- v2.0.0
  - all variants look like hard stop immediate

laurensvalk, 2023-04-29:
Stop() is instantaneous; the motors just stop controlling and coast, as if unplugging the cable. If they have a lot of speed to begin with, they’ll spin for a bit more due to inertia.

To decelerate to standstill at a controlled rate, you could drive(0, 0) and wait for the speed to become close to zero. Is that what you are asking? This is no different than drive(300, 0) elsewhere in your program, just with a different target speed.

If this is a very common request, we could add a dedicated method called decelerate which will essentially do as described above.

It sounds like you feel that one of the cases you describe is unexpected. Could you add a video of what it does, and describe what you expected it to do? Thanks!

kai-morich, 2023-04-30:
My lego team is using ev3 with v3.2.3 and faced the strange **overshoot** effect described above, thats the main reason why I isolated a testcase and opened the issue. Due to the **overshoot** effect it looked like the next command is ignored when using `DriveBase` and they started wait() and other workarounds like running motors individually.

If others do not have problems with stop and decelerate, this issue could be closed. 
We will stick to that version for the current WRO season as we have a workaround.

laurensvalk, 2023-04-30:
6 cases are described, but I'm still not sure which one is wrong. Which one has the unexpected overshoot? :)

kai-morich, 2023-05-01:
I tried to create a video, but the issue is a bit shy. Today it hardly shows up.

The issue only happens in case 3: with `stop()` but w/o `wait()` on ev3 and pybricks micropython v3.2.x
there is no issue with other devices: I tested spike and ev3 with build from #1035 and v2.0.0

At the point where it should reverse direction, it instead accellerates for 10 centimeters, then decelerates and reverses direction.

Looks like a buffer overflow or the PID controller being in some weird state if the sign of the direction changes.

laurensvalk, 2023-05-02:
> The issue only happens in case 3: with stop() but w/o wait() on ev3 and pybricks micropython v3.2.x
there is no issue with other devices: I tested spike and ev3 with build from https://github.com/pybricks/support/issues/1035 and v2.0.0

Thanks for taking the time to test all this, I appreciate it.

The original program above was using the SPIKE Prime Hub, which is why I wasn't sure where the problem was.

But if this is happening only on EV3 V3.2.x, I think this might just be the same as #1035.

kai-morich, 2023-05-02:
#1035 happens only with v3.3.* 
this issue happens only with v3.2.*, so they do not look related.

The good case is on ev3 v.3.2.* w/o stop an w/o wait. Other ev3 versions and other devices have the same behavior of basically immediately changing the direction, regardless if stop is used or not.

At bad case 1 it's a bit difficult to see the transition from orange to red, where it should change direction. Due to the short wait and _green_ in bad case 2 it's more obvious.

https://user-images.githubusercontent.com/25850372/235731979-438b9378-30fb-48ef-bcc8-997489c2f17a.mp4

https://user-images.githubusercontent.com/25850372/235732002-9a72ae75-0d98-46b3-9ee7-507d81932b61.mp4

https://user-images.githubusercontent.com/25850372/235732026-b550ea9c-45f2-40d1-821c-10402c01fd1a.mp4

laurensvalk, 2023-05-02:
> https://github.com/pybricks/support/issues/1035 happens only with v3.3.*
> this issue happens only with v3.2.*, so they do not look related.

So if we release v3.3 and we fix #1035, everything is good?

kai-morich, 2023-05-02:
yes, that's my preferred solution

closed 2023-05-02 by kai-morich

laurensvalk, 2023-05-02:
Thanks for investigating! I will focus on #1035 and report there :)