pybricks/support #1060: [Question] Will there be support for thirdparty devices like Sbrick?

Xayer, 2023-05-03:
**Question**
Will there be support for thirdparty devices like Sbrick?

I found the following documentation for their BLE protocol here:

https://social.sbrick.com/custom/The_SBrick_BLE_Protocol.pdf

**Context**
I am trying to use my SBricks in collaboration with other PU units.
I have been looking into the [Brick Automation Project](https://github.com/Cosmik42/BAP), which is able to control different kinds of BLE enabled devices together. The biggest problem with that solution is that it's Windows only - and requires a central hub (PC), to receive and transmit the signals, leading to increased latency.

I was therefore wondering if it's possible to get the pybricks software to work on the SBrick - to enable support for the other Power Functions systems, without adapter cables.

BertLindeman, 2023-05-03:
I didn't think this through properly, but it would enable control of PowerFunctions devices.
And THAT sounds very good.

laurensvalk, 2023-05-03:
Much appreciated @BertLindeman! That’s the way to make things happen. 😄

Xayer, 2023-05-03:
> I didn't think this through properly, but it would enable control of PowerFunctions devices. And THAT sounds very good.

This is exactly the kind of answer I was hoping for.

> Hi Team,
>
> There are 4 sbricks in my inventory. And currently the > newest LEGO hubs can "talk" Bluetooth BLE. 
> With pybricks those can be programmed with micropython. 
> It would be great if pybricks could communicate with (updated?) sbick firmware. 
> So pybricks could control PowerFunctions devices.

> Would there be an opportunity in that?
> I think so.

> Kind regards,
> Bert

My next question then becomes, why is the new firmware required if we have the BLE protocol that was linked above?
Is it to install micropython on the Sbrick itself?

Is there a way for us to support this? Did lego make the happen with pushback from the community?

laurensvalk, 2023-05-03:
A new firmware should not be required if there is a BLE protocol. Since you mentioned third party support in general, the case of firmware is more relevant for controllers like the [BuWizz](https://buwizz.com/shop/buwizz-3-0-pro/) controller, which is why we mentioned it.

> Is there a way for us to support this?

When we consider new features, we usually try to prioritize those features that users want most.

For third-party device support, it is a bit more subtle.

In practice, the majority of users are happy to spend $50--$150 on hardware accessories but precisely $0 on open source software (of course this is not the case for everyone!). So as you can perhaps imagine, we are cautious about spending a lot of time on it if there is no support from the relevant manufacturer.

Xayer, 2023-05-04:
> But as you can perhaps imagine, we are cautious about spending a lot of time on it if there is no support from the relevant manufacturer, when they would be the main beneficiaries of our efforts

Agreed. I have not looked into the codebase yet myself, but I would imagine that there's some sort of abstraction layer for each hub - I am wondering if the community is able to contribute (given they of course are software developers) to such an abstraction layer for the parts they have on hand?

I myself own the two types of Sbrick, and a few people from my LUG have a buwizz, and almost everything else you can imagine.

dlech, 2023-05-04:
In addition to someone just contributing the changes, we also need someone to be willing to maintain it for the foreseeable future. Pybricks is still under heavy development and each change could potentially break things unintentionally. So we would need someone to regularly test changes and make sure everything is fixed before each release.

laurensvalk, 2024-02-27:
Since the relevant manufacturers haven't gotten in touch and no maintainer is currently available for this, I think we can close this issue.

closed 2024-02-27 by laurensvalk