pybricks/support #1064: [Feature] Ideas to support user configurable menu on Spike Prime Hub

laurensvalk, 2023-05-08:
**Is your feature request related to a problem? Please describe.**
FLL teams would like to have an easy way to have an on-screen menu.

Here are some ideas.

### 1. Document how to make a Python program that does this

Something like https://github.com/pybricks/support/issues/861

### 2. Pybricks Code UI script template

Generates a script just like [`#1`](https://github.com/pybricks/support/issues/861) as one part of the "Create a new file" dialog, with dropdowns like this:

```

[F ▼]   [   fly_mission.py    ▼  ]
[D ▼]   [   drive_mission.py  ▼  ]
[Z ▼]   [   zigzag.py         ▼  ]
```

Single uppercase letters like this seems practical and easy to remember at competition day too, but any other number/visual could work.

### 3. Like `#2`, but download it right away

Instead of generating a template script, it just downloads the "menu" program. Same dropdown UI as above except:
- the button would say "download" instead of "create"
- Not part of new file dialog. Instead opened from here:

[image]

### 4. Firmware side solution

See #139. You select a slot on the hub, and Pybricks Code downloads it there.

It is not entirely clear how this should work since there is no file system.

[image]

### Additional notes.

- With option 1, 2, and 3, it is still just a program, so you need to press the start button to start it, then select a program.
- Making it like a `boot.py` could solve that, but then it not clear how to exit to allow downloading a new program. And one extra click shouldn't be too bad.

laurensvalk, 2023-05-08:
I updated "option 1" with the idea of having a `hub_menu` function available.

This would make it the simplest of all the options, and still quite easy to use and powerful. No changes to Pybricks Code or firmware required.

MonongahelaCryptidCooperative, 2023-05-08:
Here is how we did it:

https://github.com/MonongahelaCryptidCooperative/FLL/

laurensvalk, 2023-05-09:
Thanks for sharing!

laurensvalk, 2023-06-29:
We've selected option **1** for now. Other solutions may be considered in the future, but this should provide the key functionality for most use cases.

Now we just need to add documentation and/or a tutorial, see https://github.com/pybricks/pybricks-api/issues/144.

closed 2023-06-29 by laurensvalk