pybricks/support #1069: [Feature] Change definition of stalled

laurensvalk, 2023-05-11:
**Current implementation**

Currently stalled is true if:

1. The proportional part of PID exceeds the (configurable) limit _and_;
2. The motor is stopped, i.e. speed below (configurable) limit _and_;
3. it has been like this for some (configurable) time.

**Suggested improvement: Drop condition 2**

Stall detection is mostly used to detect over load **in order to stop**. We don't need to wait until we are physically stopped to test this.

Doing this makes the stall detection more sensitive, and it makes `run_until_stalled` work better. We can use a configurable torque limit (test 1 above) instead of a duty limit. This means you can run at any speed and still safely detect soft endpoints. With `duty_limit` today, this doesn't work because capping it at say 30% means you can't run fast.

I think it is still be consistent with our current definition:

[image]

**Drawbacks**
Maybe this is no longer called "stalled" this way, since it may not be fully stopped?

dlech, 2023-05-11:
Instead of changing the definition, can we just add a new `run_until_overload(speed, torque_limit, then) -> int` and `overload() -> bool` and deprecate stalled?

I think an overload condition is also useful for discovering that you are trying to run at a speed that the motor is not physically capable of.

laurensvalk, 2023-05-12:
Part of this is getting mixed in with https://github.com/pybricks/pybricks-micropython/pull/166 to make some progress.

I need to pull out the motor servo stuff, then rebase that PR.

laurensvalk, 2023-05-12:
Yes, as mentioned something like `overload` could work, I'm just not sure what to do with `stalled` as having both would be somewhat costly.

I'm wondering if it is acceptable to deprecate `stalled` by just aliasing it to the new variant. Old scripts would run, but with slightly different behavior.

Although pbio-derivatives will probably want to keep `stalled` available as well...

BertLindeman, 2023-07-25:
Maybe not 100% on this item, but as a simple user I would expect **True** as the result of ```motor.stalled()``` after running ```motor.run_until_stalled(500)```.

Currently running this on ```('primehub', '3.3.0b8', 'v1.20.0-23-g6c633a8dd on 2023-07-07')```
```
    print("angle before closing", gripper.angle())
    gripper.run_until_stalled(500)
    print("gripper motor stalled ", gripper.stalled())
    print("angle after ", gripper.angle())
```  
 which shows:
```
angle before closing 83
gripper motor stalled  False
angle after  478
```

laurensvalk, 2023-07-25:
By default, `run_until_stalled` uses `Stop.COAST`. When it coasts, it is no longer stalled.

If you change it to `Stop.HOLD`, you should find it still stalled.

BertLindeman, 2023-07-25:
It does not.
Not with V3.2.3 and not with v3.3.0b8.
Should I make another discussion to not pollute this one further?