pybricks/support #1070: Programs stops randomly

DaneelRO, 2023-05-12:
Hello everyone, 

I am working on a project involving 3 hubs; one acting as a Masterhub which broadcast information to the other two. It's an articlated arm. Up to now, I was able to write a program for each hub in order to calibrate all motors and, then, receive an input and react accordingly. Unfortunately, I am experincing an issue consisting in that one or some of the programs stop for no reason, leaving no message. They just stop. In the case the the Master hub stops (the only one that maintains the connection with the PC), the program stops and then the connection is also lost. When it happens, the hub's display switches from the "tunning circle" to the stable square 3x3. Then I have to re-start them, which is particularly annoying because of the required calibration. 

MasterHub: Gets the input, makes the calculations required for the angles of the shoulder, elbow and wrist and controls the rotation of the shoulder (4 motors). It has one sensor that allows the calibration.
ReceiverHub01: Controls the the elbow (2 motors), wrist (vertical and rotation, one motor each) and the claw (open/close, une motor). It has one sensor that allows calibration of the wrist. 
ReceiverHub02: Controls the shoulder (vertical, 4 motors). 

Any idea why the programs are stopping?

So, the programs that I am running are:

For the Master hub:
`

```
> from pybricks.hubs import InventorHub
> from pybricks.pupdevices import Motor, ColorSensor, UltrasonicSensor, ColorDistanceSensor
> from pybricks.parameters import Button, Color, Direction, Port, Side, Stop
> from pybricks.robotics import DriveBase
> from pybricks.tools import wait, StopWatch
> from pybricks.experimental import Broadcast
> from umath import asin, acos, atan, degrees
> 
> radio = Broadcast(topics=["INPUT"])
> 
> hub = InventorHub()
> BaseA = Motor(Port.A, Direction.COUNTERCLOCKWISE, ([1, 20], [12,60]))
> BaseB = Motor(Port.B, Direction.CLOCKWISE, ([1, 20], [12,60]))
> BaseC = Motor(Port.C, Direction.CLOCKWISE, ([1, 20], [12,60]))
> BaseD = Motor(Port.D, Direction.COUNTERCLOCKWISE, ([1, 20], [12,60]))
> Sensor = ColorDistanceSensor(Port.E)
> 
> radio = Broadcast(topics=["INPUT"])
> CalClaw = 0
> 
> info2=int(0)
> info6=int(0)
> 
> print("Calibrate base? (Yes=LEFT, No=RIGHT)")
> pressed = []
> while not any(pressed):
>     wait (10)
>     pressed = hub.buttons.pressed()
> if Button.LEFT in pressed:
>     calibration = 0
>     BaseA.run(10)
>     BaseB.run(10)
>     BaseC.run(10)
>     BaseD.run(10)
>     dist=0
>     while calibration < 1:
>         dist= Sensor.distance()
>         print(dist)
>         if dist < 50 :
>             wait(200)
>             BaseA.brake()
>             BaseB.brake() 
>             BaseC.brake() 
>             BaseD.brake()
>             wait(1000)
>             BaseA.run_angle(10, -72, Stop.HOLD, wait=False)
>             BaseB.run_angle(10, -72, Stop.HOLD, wait=False)
>             BaseC.run_angle(10, -72, Stop.HOLD, wait=False)
>             BaseD.run_angle(10, -72)
>             BaseA.reset_angle(0)
>             BaseB.reset_angle(0)
>             BaseC.reset_angle(0)
>             BaseD.reset_angle(0)
>             wait(1000)
[... 257 lines cut ...]
              Shoulder2R.brake()
              Shoulder1L.brake()
              Shoulder2L.brake()
              Shoulder1R.reset_angle(0)
              Shoulder2R.reset_angle(0)
              Shoulder1L.reset_angle(0)
              Shoulder2L.reset_angle(0)
              CalibrationShoulder =1
              hub.light.blink(Color.GREEN, [500, 500])
              wait(3000)
              print("Calibration completed")
  if Button.RIGHT in pressed :
      CalibrationShoulder = 1
      print("Calibration skipped")
  
  while CalibrationShoulder == 1:
      info = radio.receive("INPUT")
      wait(100)
      if info :
          Angle = info[4]
          Shoulder1R.run_target(1000,Angle,wait=False)
          Shoulder2R.run_target(1000,Angle,wait=False)
          Shoulder1L.run_target(1000,Angle,wait=False)
          Shoulder2L.run_target(1000,Angle)`
```

dlech, 2023-05-12:
> the program stops and then the connection is also lost.

This sounds like the hub is crashing. 

We have a new alternative BLE broadcast implementation we try trying out. You can get the firmware for it at https://github.com/pybricks/pybricks-micropython/pull/158#issuecomment-1509139967

You will have to adjust your programs a bit for the different API. Let us know if it works.

```python
from pybricks.hubs import InventorHub
from pybricks.pupdevices import Motor, ColorSensor, UltrasonicSensor, ColorDistanceSensor
from pybricks.parameters import Button, Color, Direction, Port, Side, Stop
from pybricks.robotics import DriveBase
from pybricks.tools import wait, StopWatch
from umath import asin, acos, atan, degrees
 
hub = InventorHub(broadcast_channel=1)
BaseA = Motor(Port.A, Direction.COUNTERCLOCKWISE, ([1, 20], [12,60]))
BaseB = Motor(Port.B, Direction.CLOCKWISE, ([1, 20], [12,60]))
BaseC = Motor(Port.C, Direction.CLOCKWISE, ([1, 20], [12,60]))
BaseD = Motor(Port.D, Direction.COUNTERCLOCKWISE, ([1, 20], [12,60]))
Sensor = ColorDistanceSensor(Port.E)
 
CalClaw = 0
 
info2=int(0)
info6=int(0)
 
print("Calibrate base? (Yes=LEFT, No=RIGHT)")
pressed = []
while not any(pressed):
    wait (10)
    pressed = hub.buttons.pressed()
if Button.LEFT in pressed:
    calibration = 0
    BaseA.run(10)
    BaseB.run(10)
    BaseC.run(10)
    BaseD.run(10)
    dist=0
    while calibration < 1:
        dist= Sensor.distance()
        print(dist)
        if dist < 50 :
            wait(200)
            BaseA.brake()
            BaseB.brake() 
            BaseC.brake() 
            BaseD.brake()
            wait(1000)
            BaseA.run_angle(10, -72, Stop.HOLD, wait=False)
            BaseB.run_angle(10, -72, Stop.HOLD, wait=False)
            BaseC.run_angle(10, -72, Stop.HOLD, wait=False)
            BaseD.run_angle(10, -72)
            BaseA.reset_angle(0)
            BaseB.reset_angle(0)
            BaseC.reset_angle(0)
            BaseD.reset_angle(0)
            wait(1000)
            calibration =1
            hub.light.blink(Color.GREEN, [500, 500])
            wait(3000)

if Button.RIGHT in pressed or calibration == 1:
    info1=int(0)
    while CalClaw == 0:
        print("Close Claw (0 to 5000, Close: Neg, Open: Pos):")
        Adjust=int(input())
        hub.ble.broadcast(Adjust, 0, 0, 0, 0)
        if Adjust == 0:
[... 220 lines cut ...]
            Shoulder2R.brake()
            Shoulder1L.brake()
            Shoulder2L.brake()
            Shoulder1R.reset_angle(0)
            Shoulder2R.reset_angle(0)
            Shoulder1L.reset_angle(0)
            Shoulder2L.reset_angle(0)
            CalibrationShoulder =1
            hub.light.blink(Color.GREEN, [500, 500])
            wait(3000)
            print("Calibration completed")
if Button.RIGHT in pressed :
    CalibrationShoulder = 1
    print("Calibration skipped")

while CalibrationShoulder == 1:
    info = hub.ble.observe(1)
    wait(100)
    if info is not None:
        Angle = info[4]
        Shoulder1R.run_target(1000,Angle,wait=False)
        Shoulder2R.run_target(1000,Angle,wait=False)
        Shoulder1L.run_target(1000,Angle,wait=False)
        Shoulder2L.run_target(1000,Angle)
```

DaneelRO, 2023-05-13:
Thanks a lot @dlech! I have installed the firmware in all the three hubs and made the appropriate modifications that you showed. 

Nevertheless, in none of the program the dashboard is printing anything. Is this a reported issue?

Many thanks!!!!

laurensvalk, 2023-05-13:
You have to use beta.pybricks.com

DaneelRO, 2023-05-13:
Great! Many thanks for the support! it runs now nice and smooth! :)

closed 2023-05-26 by dlech