pybricks/support #1072: how to install the numpy package on ev3 now

Nikolay-Zagrebin, 2023-05-15:
**Question**
how to install the numpy package on ev3?

**Context**
previously, it was possible to install the command
robot@ev3dev:~$ sudo apt-get install python3-numpy

an error is coming out now

Nikolay-Zagrebin, 2023-05-15:
maybe I asked a question in the wrong section. I probably should have gone to ev3dev

dlech, 2023-05-15:
Yes, please post the exact error message on the ev3dev issues. I'm guessing you forgot to run `sudo apt update` first or the EV3 is not connected to the Internet.

closed 2023-05-15 by dlech

Nikolay-Zagrebin, 2023-05-16:
I figured it out, Debian moved everything to the archive.
To solve the problem, add a new line to the sources.list file
deb http://archive.debian.org/debian stretch main contrib non-free