pybricks/support #1086: Technic Hub not broadcasting when disconnected from computer

laurensvalk, 2023-05-28:
> I tested the new example programs included in the documentation of hub.ble.broadcast / observe  functions included in latest 2.2.0-beta.4 release. I'm really happy to see it finally implemented, thank you for that!
While connected to Pybricks API, everything goes well, just as expected. But on Technic Hub there is a consistent problem of not broadcasting any data when the code is run directly from the hub (after disconnecting from pc). The other Techic Hub acting as receiver keeps receiving ble signal when disconnected from pc and code run from directly hub. When code is adapted to Inventor Hub and started directly from Hub everything goes fine. So the problem occurs only on Tecyhnic Hub and only in broadcasting mode. I tested on several Technic Hubs as broadcaster and various hubs as receiver, and the problem is 100% persistent.

> Another observation: there is much less latency in receiver Technic Hub motor folowing the broadcaster Hub's motor movements when the broadcaster is Inventor Hub (it is close to 100ms as observed). It seems as Technic Hub has much more than 100ms intervals (rather 800-1000ms as observed) between data transmission session.

_Originally posted by @vardayzsolt in https://github.com/pybricks/pybricks-micropython/issues/158#issuecomment-1566138121_

laurensvalk, 2023-05-28:
When it _is_ connected to the computer, broadcasting with Technic Hub will have some additional latency.

It should go back to normal when the program is run from the hub without a PC connection. But if I understand you correctly it does not work for you at all in that case. Thanks for reporting. We'll want to fix that.

It does sound quite similar to @nstrijbosch's findings.

vardayzsolt, 2023-06-01:
[hthtest1.zip](https://github.com/pybricks/support/files/11628620/hthtest1.zip)
Here you have in a .zip file all three codes, adopted for single motor from example scripts in beta.pybricks documentation.
The receiver is always a TechnicHub, the sender is either a TechnicHub (stops working when disconnected from PC) either an InventorHub (works just fine when disconnected from PC). I tried to run the code on 3 different TechnicHubs, all of them flashed with latest beta firmware (I suppose that with older firmware this code cannot run anyway when connected to PC due to inexistent ble.broadcast method)

dlech, 2023-06-01:
Thanks. I seems to be able to reproduce the problem now.

closed 2023-06-01 by dlech