pybricks/support #1088: [Bug] Motors strart blocking

Tear4Pixelation, 2023-05-29:
**Describe the bug**
What is the problem?
So I made myself a new Class - see how to reproduce - and rarely, when I put it after for example `robot.turn(90) robot.stop()`, the motors just start blocking.
**To reproduce**
Steps to reproduce the behavior:
My class:
`class Motorctrl:
    def __init__(self, left_motor, right_motor, dia, wheel_dist):
        self.left_motor = left_motor
        self.right_motor = right_motor
        self.dia = dia
        self.wheel_dist = wheel_dist
    def run(self, speed_left, speed_right):
        self.left_motor.run(speed_left)
        self.right_motor.run(speed_right)
    def stop(self):
        self.left_motor.stop()
        self.right_motor.stop()
    def reset_dist(self):
        self.left_motor.reset_angle(0)
        self.right_motor.reset_angle(0)
    def dist(self):
        return self.dia * math.pi * (self.left_motor.angle() + self.right_motor.angle()) / 2 /360
    def lv(self, dist, stopcolor = None, speed = 450, times = 0.98):  
        sp = speed * 0.98
        self.reset_dist()
        while dist > self.dist():
            sp = sp * 0.98
            wait(20)
            if cs_left.color() == stopcolor and cs_right.color() == stopcolor: # stop at end of line and double-check
                self.stop()
                for i in range(30):
                    wait(20)
                    if not cs_left.color() == stopcolor and not cs_right.color() == stopcolor:
                        continue 
                if not cs_left.color() == stopcolor and not cs_right.color() == stopcolor:
                    continue
                else:
                    break
            if cs_left.color() == Color.WHITE and cs_right.color() == Color.WHITE or cs_left.color() == Color.BLACK and cs_right.color() == Color.BLACK:
                self.run(speed, speed)
                sp = speed * 0.98
            elif cs_right.color() == Color.BLACK and cs_left.color() == Color.WHITE:
                self.run(speed, sp) # Turn right
            elif cs_right.color() == Color.WHITE and cs_left.color() == Color.BLACK:
                self.run(sp, speed) # Turn left
            elif not cs_right.color() in [Color.BLACK, Color.WHITE]:
                self.run(sp, speed)
            elif not cs_left.color() in [Color.BLACK, Color.WHITE]:
                self.run(speed, sp)            
        self.stop()

    def lv_whiteblack(self, dist):  
        self.reset_dist()
        sp = 400 * 0.98
        while dist > self.dist():
            sp = sp * 0.98
            wait(20)
            if cs_left.color() == Color.BLACK and cs_right.color() == Color.BLACK: # stop at double-black  
                break
            elif cs_left.color() == Color.WHITE and cs_right.color() ==  Color.BLACK:
                self.run(400, 400)
                sp = 400 * 0.98
            elif cs_left.color() == Color.BLACK or (cs_left.color() == Color.WHITE and cs_right.color() == Color.WHITE):
                self.run(sp, 400) # Turn left
            elif cs_right.color() == Color.WHITE and not cs_left.color() in [Color.BLACK, Color.WHITE]:
                self.run(400, sp) # Turn right   
        self.stop()
    def drive_untl(self, speed_left, speed_right, color_left = None, color_right = None):
        if color_right is not None and color_left is not None:
            while not (cs_right.color() == color_right and cs_left.color() == color_left):
                self.run(speed_left, speed_right)
        elif color_right is not None:
            while not cs_right.color() == color_right:
                self.run(speed_left, speed_right)
        elif color_left is not None:
            while not cs_left.color() == color_left:
                self.run(speed_left, speed_right)
        self.stop()
    def lv_straight(self, dist, stopcolor = None, speed = 450, times = 0.985):  
        sp = speed * 0.98
        self.reset_dist()
        while dist > self.dist():
            sp = sp * times
            wait(20)
            if cs_left.color() == stopcolor and cs_right.color() == stopcolor: # stop at end of line and double-check
                self.stop()
                for i in range(30):
                    wait(20)
                    if not cs_left.color() == stopcolor and not cs_right.color() == stopcolor:
                        continue 
                if not cs_left.color() == stopcolor and not cs_right.color() == stopcolor:
                    continue
                else:
                    break
            if cs_left.color() == Color.WHITE and cs_right.color() == Color.WHITE or cs_left.color() == Color.BLACK and cs_right.color() == Color.BLACK:
                self.run(speed, speed)
                sp = speed * times
            elif cs_right.color() == Color.BLACK and cs_left.color() == Color.WHITE:
                self.run(speed, sp) # Turn right
                times = times + 0.01
            elif cs_right.color() == Color.WHITE and cs_left.color() == Color.BLACK:
                self.run(sp, speed) # Turn left
                times = times + 0.01
            elif not cs_right.color() in [Color.BLACK, Color.WHITE]:
                self.run(sp, speed)
                times = times + 0.01
            elif not cs_left.color() in [Color.BLACK, Color.WHITE]:
                self.run(speed, sp)
                times = times + 0.01            
        self.stop()
    def curve(self, angle, lft = 'auto', speed = 500):
        rot_angl = self.wheel_dist / (2 * self.dia) * 4 * angle  
        if lft == 'auto':
            if angle > 0:
                self.left_motor.run_angle(500, rot_angl)
            else:
                self.right_motor.run_angle(500, -1*rot_angl)
        elif lft:
            self.left_motor.run_angle(500, rot_angl)
        else:
            self.right_motor.run_angle(500, rot_angl)
`

**Expected behavior**
The robot should continue

**Screenshots**
There is a saying that a picture is worth a 1000 words. Screenshots really help to identify and solve problems.
Wheels is the name for the Motorctrl class, Robot isthe drive base and `wheels.drive_untl()` is driving with both motors at given speed until one or two color sensor reach a specific colour as you can see in the class.
The motor is trying to run (maybe you can hear it), but it won't move. Its like something is commanding one motor to go in two different directions at once.
https://github.com/pybricks/support/assets/104512895/72970b06-00d3-4f70-bfd7-7d170a8bde18

laurensvalk, 2023-05-29:
Is this using the standard EV3 MicroPython version?

If you have a shorter script that reproduces the problem that would be great.

Not sure if related, but it also helps to have a short wait in your loops:

while some_condition:
    motor.run(...)
    wait(10)  # <<--- Putting a short wait here gives your program time to do other things.

Tear4Pixelation, 2023-05-29:
> ```python
> while some_condition:
>     motor.run(...)
>     wait(10)  # <<--- Putting a short wait here gives your program time to do other things.
> ```

I have now done that, but it did not help :(
I suspect it has something to do with the interference between the drive base and motorcrtl, because I can hear that the motors inside the motors are working, but the axe just does not turn.

laurensvalk, 2023-05-29:
Ah, are you using the motors individually, as well as in the Drive Base at the same time?

If you do, you'll want to make sure to call `robot.stop()` before using the motors again, and likewise call `left_motor.stop()` and `right_motor.stop()` before you use the drive base again.

In newer versions this is handled automatically, but you can probably solve this by doing this yourself.

Tear4Pixelation, 2023-05-29:
I won't close this issue because it does not feel like the actual issue is fixed, but I started the program again from the start, and then everything worked again (I have a system that lets me skip phases of my program)

Tear4Pixelation, 2023-05-29:
> Ah, are you using the motors individually, as well as in the Drive Base at the same time?
>
> If you do, you'll want to make sure to call `robot.stop()` before using the motors again, and likewise call [...]

You can look at the video, I use robot.stop() or here
>So I made myself a new Class - see how to reproduce - and rarely, when I put it after for example robot.turn(90) >robot.stop(), the motors just start blocking.

Tear4Pixelation, 2023-05-29:
> Ah, are you using the motors individually, as well as in the Drive Base at the same time?
>
> If you do, you'll want to make sure to call `robot.stop()` before using the motors again, and likewise call [...]

From what version is thin handled? I need to get it xD

laurensvalk, 2023-05-29:
> > In newer versions this is handled automatically, but you can probably solve this by doing this yourself.
> 
> From what version is thin handled? I need to get it xD

I really don't recommend changing versions since you say:

> our competition with the robot is soon

The EV3 version hasn't had a stable release in a while, as we've been prioritizing all the other hubs.

Tear4Pixelation, 2023-05-29:
> I really don't recommend changing versions since you say:
>
>
> The EV3 version hasn't had a stable release in a while, as we've been prioritizing all the other hubs.

Ok, I might give it a go at some point, but I'll wait if you say so. Thank you a lot for your quick help, the pyricks support is great. Keep it up!

dlech, 2025-07-13:
Closing this since we're dropping support for ev3dev in favor of a new EV3 solution.