pybricks/support #1102: Alternate to `input` and `usys.stdin` on Move hub

dlech, 2023-06-06:
**Is your feature request related to a problem? Please describe.**
This has come up several times recently (#1099, #1085, #529). There is no easy way to get input from a computer to a move hub. 

**Describe the solution you'd like**
A function similar to `usys.stdin.read()` that plays nice with the new async run loop with minimal implementation size.

**Describe alternatives you've considered**
* The new broadcast/observe functionality is not a complete solution since it can't be used on macOS and requires experimental BlueZ features to be manually enabled on Linux (or elevated privileges to directly poke HCI).
* We have also proposed an API for wrapping the Nordic UART service, but the Move hub currently doesn't have anywhere near enough free flash memory to implement this.
* We could implement a very minimal `mp_hal_readline()` (e.g. backspace wouldn't work) and use the existing builtin `input()` function. However, since this prints in addition to getting input, it would not be ideal for use in a program with an async run loop where we need it to be awaitable rather than blocking.

laurensvalk, 2023-06-06:
We've previously had a non-blocking `getchar` that seemed to work well for most people.

dlech, 2023-06-06:
:+1: That is probably as simple as it gets.

closed 2023-06-12 by dlech