pybricks/support #1117: Unable to pair[Question]

PaulLBos, 2023-06-16:
**Question**
I am unable to pair any of my hubs I have been using Pybricks for about 2 years but since i have updated cannot reach any of the hubs I have followed the instructions to load Pybricks on to the hubs but it can not see them my computer can pair with the head phones and the usb to bluetooth worked with Pybricks before. I am on windows 11 so it may be one of their updates causing the problem

**Context**
to continue using Pybricks on my projects

**Screenshots**
There is a saying that a picture is worth a 1000 words. Screenshots really help to identify and solve problems.

dlech, 2023-06-16:
I assume when you say "using Pybricks", you mean "using Pybricks Code at https://code.pybricks.com", is that correct?

When you say "pair" I assume that you mean you are clicking the Bluetooth button in Pybricks Code to connect to a hub and you are *not* using the operating system Bluetooth settings to try to connect to the device, is that correct?

If my assumptions are correct, then I think your guess is correct and you could try rolling back Windows updates or try a different driver version for your Bluetooth adapter to see if that fixes the problem.

PaulLBos, 2023-06-21:
Now sorted. Failed to notice in the instructions that you have to click on the hub in the pairing window to make the pair button active.

closed 2023-06-21 by dlech