pybricks/support #1123: [Question] Turn on Bluetooth when Click Bluetooth button on Brick

ttn-c4plus, 2023-06-22:
**Question**
Hello, the most important thing is the Brick always turn on Bluetooth and ready for connect when it on
This is very dangerous when we join competition, because some bad guys can connect to the Robot at competition area.
How can I turn on Bluetooth when click Bluetooth button on brick?

dlech, 2023-06-22:
This is not currently possible. It was discussed recently at https://github.com/orgs/pybricks/discussions/1089 and someone made a quick hack for their own use and shared it.

But it sounds like something we should consider including in the default firmware.

ttn-c4plus, 2023-06-23:
Yeah, if it's going to true, it will be awesome..

laurensvalk, 2023-06-30:
Here's how it might work from a user perspective.

- Bluetooth light is activated to indicate Bluetooth state.
    - Light is blue means Bluetooth is on, light off means off.
    - Connection state is unchanged, we'd still use the center light for that to be consistent with other hubs.
- Pressing the Bluetooth button toggles it on/off.
    - This can only be done in idle state when program is not running
    - When program is running, button is used as normal button, It won't toggle BT state and light.
    - Bluetooth is on by default. 
    - The state could be made persistent by storing it in a section of the user data field.

nkarstens, 2023-09-02:
I have an implementation of this in a draft pull request here: pybricks/pybricks-micropython#195. It mostly works, but I'm having trouble with the persistent storage part of it. If someone could take a look it would be appreciated!

nothingtodowu, 2024-04-15:
Hi Karstens,

I had clone and build with this image, but when I download it to primehub, when I press bluetooth button, nothing happen. What should I do to turn off/on the bluetooth? Thanks a lot.

https://github.com/nkarstens/pybricks-micropython/tree/bluetooth-toggle

nothingtodowu, 2024-04-17:
Hi Karstens,

Thanks a lot. It fiexed. You have done a great job.

nkarstens, 2024-04-20:
@nothingtodowu Thanks, glad it worked out for you!

laurensvalk, 2024-10-05:
This has been included in the firmware now. It is available as a nightly build and will be available in the upcoming beta release.

closed 2024-10-05 by laurensvalk