pybricks/support #1131: [Bug] TechnicHub with L (88013) Motor

tomijah, 2023-07-01:
Seems like L 88013 motors are not being recognized. I tried it on City, Technic, Inventors Hubs. 
I tested all ports in Technic Hub with 3 different 88013 motors. I also tried to mix up the ports (like connect to A and indicate B in code). Every time I'm getting 

[image]

My code:
```
from pybricks.hubs import TechnicHub
from pybricks.pupdevices import Motor
from pybricks.parameters import Button, Color, Direction, Port, Side, Stop
from pybricks.robotics import DriveBase
from pybricks.tools import wait, StopWatch

hub = TechnicHub()
mot = Motor(Port.A)

mot.run(900)
wait(2000)
```

**To reproduce**
1. Connect 88013 to any port of the TechnicHub
2. Run the above program

**Expected behavior**
The motor should run for 2 seconds

tomijah, 2023-07-01:
This is just one example of the issues I encountered after installing v3.3.0b7. I won't be ticketing all of them right away just in case they could have the same cause (e.g. the latest refactors https://github.com/pybricks/pybricks-micropython/pull/174)

KWSmit, 2023-07-01:
I just tested it on one of my Technic Hubs with motors 88018 and 88008, ColorDistanceSensor and Light. Everything seems to work fine when motor or sensor connected to Ports A, C or D. Only port B gives the same result as mentioned by Tomasz for both motors and the ColorDistanceSensor: in REPL the statement `m = Motor(Port.B)` gives `OSError: [Errno 19] ENODEV:` etc. With the lights there is no error, but lights won't turn on. When using one of the other ports there is no problem and everything seems to work fine...

tomijah, 2023-07-01:
For me, 88018 and 88008 doesn't work on port A but work on any other. 88013 consistently doesn't work on any of the ports.
Also the same case with Light (doesn't crash but lights won't turn on)

laurensvalk, 2023-07-01:
Thanks for reporting! We changed the port interface extensively so it's possible something slipped through.

If anyone has a prime hub or inventor hub to repeat the tests that would be useful.

I'll try reproducing it on all hubs on Monday.

laurensvalk, 2023-07-01:
I can reproduce this. This does not seem to affect the Prime Hub (where most testing was done).

On Technic Hub, UART sensors are giving this error while syncing up:

```Data error: Received INFO for incorrect mode```

And then the detection process for that port gets confused. So at first glance, this does not seem unique to a particular port, it just appears that way if something was plugged in and then removed.

KWSmit, 2023-07-01:
Just tested it briefly on Inventor Hub. Mine is build now as Gelo and all motors and sensors work fine on all ports.

tomijah, 2023-07-01:
### InventorHub: ###
Motors 88018 and 88008 - work on all ports
Motor 88013 - doesn't work at all
ColorSensor 45605 - work on all ports
Light 88005 - on all ports: no error, no light
45604 UltrasonicSensor - works on all ports (including lights)

laurensvalk, 2023-07-01:
Thank you both for the fast responses! We should be able to fix this in the coming week.

laurensvalk, 2023-07-03:
The Technic motor and Light were not working on any hub in this release, which has now been fixed. These were small independent mistakes made during a large code overhaul we did recently.

The next step is to fix device synchronization on Technic Hub.

closed 2023-07-03 by laurensvalk

laurensvalk, 2023-07-03:
@KWSmit, @tomijah, to try this out, get the firmware from https://nightly.link/pybricks/pybricks-micropython/workflows/build/master and install it like this: https://pybricks.com/install/technic-boost-city/#installing-the-latest-build-advanced

BertLindeman, 2023-07-03:
Confusion here, Laurens.

Installing the CI-firmware you point to above shows:
```('technichub', '3.3.0b7', 'v1.20.0-23-g6c633a8dd on 2023-07-03')```

Basic 3.3.0b7 firmware shows:
```('technichub', '3.3.0b7', 'v1.20.0-23-g6c633a8dd on 2023-06-30')```

Only the date changed. I would expect "the git id" to show something like `ef4ae70`

Probably my ignorance . . . 

Bert

laurensvalk, 2023-07-03:
What are you seeing on this screen?

[image]

tomijah, 2023-07-03:
Just tested on Technic and Prime hubs and 88013 motor and 88005 Light work nicely.

BertLindeman, 2023-07-03:
> What are you seeing on this screen?
> 
> [image]

[image]

Did another try and got this version:
```('technichub', '3.3.0b7', 'v1.20.0-23-g6c633a8dd on 2023-07-03')```

laurensvalk, 2023-07-04:
**Use these links to test version where motors/sensors also work when already plugged in**

* [cityhub-firmware-build-2962-git5f1c12d1.zip](https://nightly.link/pybricks/pybricks-micropython/actions/artifacts/784653253.zip)
* [essentialhub-firmware-build-2962-git5f1c12d1.zip](https://nightly.link/pybricks/pybricks-micropython/actions/artifacts/784653254.zip)
* [movehub-firmware-build-2962-git5f1c12d1.zip](https://nightly.link/pybricks/pybricks-micropython/actions/artifacts/784653255.zip)
* [mpy-cross.zip](https://nightly.link/pybricks/pybricks-micropython/actions/artifacts/784653256.zip)
* [nxt-firmware-build-2962-git5f1c12d1.zip](https://nightly.link/pybricks/pybricks-micropython/actions/artifacts/784653257.zip)
* [pr_number.zip](https://nightly.link/pybricks/pybricks-micropython/actions/artifacts/784653258.zip)
* [primehub-firmware-build-2962-git5f1c12d1.zip](https://nightly.link/pybricks/pybricks-micropython/actions/artifacts/784653259.zip)
* [pybricks-micropython-build-2962-git5f1c12d1.zip](https://nightly.link/pybricks/pybricks-micropython/actions/artifacts/784653260.zip)
* [technichub-firmware-build-2962-git5f1c12d1.zip](https://nightly.link/pybricks/pybricks-micropython/actions/artifacts/784653261.zip)

laurensvalk, 2023-07-04:
These links should fix this open issue as a whole.

If anyone wants to give this a go, we may be able to release a new beta for everyone by the end of the week :smile: 

We'll also do some more testing ourselves across a range of hubs/sensors/motors.

BertLindeman, 2023-07-04:
Confirmed OK on TechnicHub.

And now with a version that I understand a bit better 😉 
```('technichub', '3.3.0b7', 'ci-build-2962-v3.3.0b7-8-g5f1c12d1 on 2023-07-04')```

KWSmit, 2023-07-04:
Confirmed OK on TechnicHub.

Motor (88008 and 88018) doesn't seem to work on CityHub, whatever I try. Light does.

Hope to try Robot Inventor later tonight...

laurensvalk, 2023-07-04:
City Hub should be fixed on https://nightly.link/pybricks/pybricks-micropython/workflows/build/master when it finishes building in 15 minutes or so. Thanks for reporting @KWSmit !

KWSmit, 2023-07-04:
@laurensvalk CityHub ok now.

I have a question: I only can run programs by using Pybricks webapp.  Running from the commandline with pubricksdev does not work: no error but program does nothing. Can this be because of new beta for firmware on hub?

To test on Robot Inventor, can I use the link to the Prime firmare?

KWSmit, 2023-07-04:
Forget my first question abou pybricksdev, it works fine now. Probably did something wrong...

dlech, 2023-07-04:
> To test on Robot Inventor, can I use the link to the Prime firmare?

Yes, they both use the same firmware.

KWSmit, 2023-07-05:
I don't have the time right now for a thourough test, but Robot Inventor seems to work fine now. Tested with Motors (88008 and 88018), Light, UltrasonicSensor and Colorsensor on different ports. Motors/sensors already connected or changing ports works well.