pybricks/support #1151: [Bug] Prime Hub broadcast does not work properly if broadcasting too fast

laurensvalk, 2023-07-09:
> here is my code:

```python
from pybricks.hubs import PrimeHub
from pybricks.pupdevices import Motor, ColorSensor, UltrasonicSensor, ForceSensor
from pybricks.parameters import Button, Color, Direction, Port, Side, Stop
from pybricks.robotics import DriveBase
from pybricks.tools import wait, StopWatch

hub = PrimeHub(broadcast_channel=10)
mode = 0

while (True):
    print('') # the problem
    if (mode == 0):
        direction = 0
        x = 0
        y = 0
        t = hub.imu.tilt()
        if (t[1] > 30):
            y = 1
        elif (t[1] < -30):
            y = -1
        
        if (t[0] > 30):
            x = -1
        if (t[0] < -30):
            x = 1
        hub.ble.broadcast(x, y)
```

> this code is for broadcasting signal to the other hub, but I realized that I can only broadcast this consistently if I connect this to my computer, and print something. otherwise, it would only send every two seconds, for about 5 pulses instead of sending signal all the time. Is there a way to make the robot work more independently? I don't want to always connect this hub to my computer when I run it.

_Originally posted by @Potato-eater in https://github.com/pybricks/support/discussions/1149_

laurensvalk, 2023-07-09:
On Prime Hub, we could add a `wait(max(0, 10 - (time_now - time_last_broadcast)` before broadcasting. This way it introduces delays only in tight loops.

Having this method (a)wait would make it consistent with the other hubs, which is probably a good thing.

laurensvalk, 2023-10-24:
The PT_THREAD for this is already set up in the right form. It probably just needs a short wait in it (but only if recently called, as per above).

laurensvalk, 2023-10-26:
I can reproduce this problem consistently as follows:

SPIKE Prime receiver with this script, running disconnected from PC:

```python
from pybricks.hubs import ThisHub
from pybricks.parameters import Color
from pybricks.tools import wait, StopWatch

hub = ThisHub(observe_channels=[12])

watch = StopWatch()

while True:
    data = hub.ble.observe(12)
    if data:
        hub.display.number(data)
    wait(10)
```

SPIKE Essential sender:

```python
from pybricks.hubs import EssentialHub
from pybricks.pupdevices import ColorLightMatrix
from pybricks.parameters import Color, Port
from pybricks.tools import wait, StopWatch

hub = EssentialHub(broadcast_channel=12)

watch = StopWatch()

while True:
    for i in range(100):
        data = i
        color = Color.RED if i % 2 else Color.GREEN
        hub.ble.broadcast(data)
        print(watch.time())
        hub.light.on(color)
        # wait(10)   <--- Needed to make it work when disconnected from PC
```

laurensvalk, 2023-10-29:
This fix had to be undone via https://github.com/pybricks/pybricks-micropython/commit/da7565e5a950120db038b2f3ea7423fa5d34a920

So we should also revert https://github.com/pybricks/pybricks-micropython/commit/cf733805e803ed0b5c19a35c776f04135fc94500 which can now hang the hub.

dlech, 2023-10-29:
It should also be possible to use the BTStack timer API instead, if needed.

closed 2023-11-04 by dlech