pybricks/support #1162: [Bug] PUPDevice does not support passive devices

laurensvalk, 2023-07-15:
### Discussed in https://github.com/orgs/pybricks/discussions/1161

<div type='discussions-op-text'>

<sup>Originally posted by **vascolp** July 15, 2023</sup>
Hello,

Version 3.3.0b8.

PUPDevice(Port.A) does not work with sensorless motors: 

- 1- Powered Up Medium Motor
- 2- Powered Up Train Motor

```
Traceback (most recent call last):
  File "pupdevice.py", line 6, in <module>
OSError: [Errno 19] ENODEV:

A sensor or motor is not connected to the specified port:

```
I was assuming that PUPDevice() is the correct object to check if the motor is sensor or sensorless... 

</div>

laurensvalk, 2023-07-15:
This is technically correct behavior, but since it was able to detect passive devices previously, I agree that we should not remove that functionality. :smile:

Thanks for posting. It is noted for fixing before the release is out.

vascolp, 2023-10-24:
I am do the opposite of that, try for Motor() and if it fails, use DCMotor().
That way I think it does not work because DCMotor() can handle motors with sensors ...

laurensvalk, 2023-10-24:
Both approaches will work. Asking it the other way, then --- Would you say it is acceptable do it in such a way, instead of doing `PUPDevice` for all devices?

After reviewing and updating the `PUPDevice` class, it seems to make sense to use it only for devices that actually do support `read` and `write`, which don't make a lot of sense for `DCMotors`.

BertLindeman, 2023-10-24:
No preference from me.

It would be nice to have a complete list (or dict) of the so called passive devices.
As example: I do not have a DCmotor so cannot test with it.
Have updated (provisional) my discover program and that now works with Poered up Light.
It uses your snippet from above.

vascolp, 2023-10-24:
I think ideally PUDevice should handle all types of devices. By its name it looks to be THE function to give info of what is attached to a port. Even considering things as simple as a light.

But I can understand the thecnical issues that prevent that and, in my case, I can live without it...

laurensvalk, 2023-10-26:
> I think ideally PUDevice should handle all types of devices.

Noted and done! Detecting will work again with the upcoming release. An updated beta should be out tomorrow.

closed 2023-10-26 by laurensvalk

vascolp, 2023-10-26:
Cool!!!!
Question: so can I assume that every device has some form if ID? That means that for new devices you will have to update the list? Or the id number is written in the device?

BertLindeman, 2023-10-26:
You **have** been productive the last week, Laurens!

Thank you so much!

Bert

laurensvalk, 2023-10-26:
For the "smart" UART devices, the ID comes from the device. New devices should work automatically as long as they follow the LEGO protocol. In practice, this "standard" changes every time a new device comes out, so sometimes we have to handle some extra cases before new sensors work.

For the passive devices like motors, there are 8 combinations based on how they are wired, so we calculate a number from there.

laurensvalk, 2023-10-26:
> You have been productive the last week, Laurens!

You should see why we have been so quiet for the past months. There has been a _lot_ going on behind the scenes and I'm super excited to show this in a few weeks if we can make it :smile: :rocket: :ship:

vascolp, 2023-10-26:
> [updated] You TWO **have** been productive the ~last~ ~week~ past months, Laurens, David!
> 
> Thank you so much!
> 
> Bert

I agree! Thank you!

laurensvalk, 2023-10-26:
Thanks! :smile:  V3.3.0b9 is out now, including the `PUPDevice` fix and several other fixes. Hopefully this is close enough to a final release, with V3.4.0a1 already around the corner :wink:

BertLindeman, 2023-10-27:
Confirmed with Poered up light.

Thanks