pybricks/support #1170: [Doc] Value range / explanation for Motor “profile”

mahalis, 2023-07-21:
**Describe the issue**
The current documentation for the [Motor](https://docs.pybricks.com/en/latest/pupdevices/motor.html) initializer states:

> A lower value means more precise movement; a larger value means smoother movement. If no value is given, a suitable profile for this motor type will be selected automatically.

It’s unclear what the units are or what a reasonable range of values are.

**Suggested improvement**
Specifying a few example values for different motor types would be helpful (it looks like a bunch of them are defined in `lib/pbio/src/motor/servo_settings.c`, ranging from 10 for the EV3 motors to 20 for some of the Technic ones), as would including this comment from `lib/pbio/include/pbio/servo.h`:

> This is a measure for "precision" (low values) versus "smoothness" (high value), expressed roughly as the allowed deviation (in degrees) from a target angle to be considered on target.

In the course of researching these suggested improvements I’ve answered the question I had… but it might be worth saving future readers that research 🙂

laurensvalk, 2023-07-21:
Thanks for mentioning this. You also get bonus points for finding the solution in the source code :smile:

laurensvalk, 2023-10-26:
Revised text:

[image]

laurensvalk, 2023-10-26:
Fixed by https://github.com/pybricks/pybricks-api/commit/e7377dff3166bcccd63a5f6b1c175d8ffe90ac83

closed 2023-10-26 by laurensvalk