pybricks/support #1176: Use functions such as run_time, run_angle or run_target for DriveBase

OverclockedToaster, 2023-07-24:
The title may be a bit confusing, but basically I want an ev3 robot with two wheels to drive for a certain amount of time/ for a certain angle. Is there any way to accomplish this behaviour? Also, what is right amount of speed for the large motors to equal 100% in EV3-G? I'm pretty new to pybricks(but not that new to python if that helps) and Github, so please apologize if this question has already been answered or mislabeled.

Thanks in advance

laurensvalk, 2023-07-24:
To find the rated speeds, print `my_motor.control.limits()`, which will give you the speed limit and acceleration as the first two values where `my_motor` is your motor in your script.

We made quite a few additions to the [DriveBase](file:///home/laurens/git-pybricks/pybricks-api/doc/main/build/html/robotics.html) recently, but those updates aren't currently planned for EV3

OverclockedToaster, 2023-07-24:
Ok, I will try that, thanks alot! Is there a way for the first problem to be solved without the updates? Also, is there any documentation on this module other than [this](https://pybricks.com/ev3-micropython/index.html)? 
Thanks for the quick reply :-)

laurensvalk, 2023-07-24:
To drive for time, you could `drive(speed, turn_rate)`, then wait for the desired duration, and then stop.

OverclockedToaster, 2023-07-24:
> To drive for time, you could drive(speed, turn_rate), then wait for the desired duration, and then stop.

That was also the solution I came to, but it seemed kinda janky to me.
> use straight() and divide the distance as needed

Ok, going to try that!
> For driving, though, isn't the intention usually to drive the robot by a given amount, instead of the wheels?

I definitely agree, but my son is trying to do the exercises (that often require angle or time specifications) in your book on the EV3 (extremely good and helpful for learning graphical programming!) with Python as he already completed it with the official Lego software years ago and is now trying to expand his Python skills.

OverclockedToaster, 2023-07-26:
Oh, ok, makes sense😄! Is there still a chance that the EV3 will be supported with Pybricks 3.3? My son now updated his EV3 to 3.3 manually, lets hope it's stable...
Btw, he tried `my_motor.control.limits()` function but Python throws a `ValueError: Invalid argument`. I think its due to some unit conflict (the documentation says that `control.limits()` outputs a value in deg/s or in mm/s). How can I solve this issue and get the motor to run at e.g. 10%?
Here's the code I ran:
```

#!/usr/bin/env pybricks-micropython

from pybricks.hubs import EV3Brick
from pybricks.ev3devices import Motor
from pybricks.parameters import Port, Stop
from pybricks.robotics import DriveBase
from pybricks.tools import wait
from time import sleep
# Initialize the EV3 Brick.
ev3 = EV3Brick()

# Initialize the motors.
left_motor = Motor(Port.B)
right_motor = Motor(Port.C)

# Initialize the drive base.
robot = DriveBase(left_motor, right_motor, wheel_diameter=43, axle_track=141)
speed = right_motor.control.limits()[0]
acceleration = right_motor.control.limits()[1]
robot.settings(straight_speed=speed, straight_acceleration=acceleration)
robot.straight(distance=1000, wait=False)

# Output:  
# File "/home/robot/Book/Test.py", line 20, in <module>
# ValueError: Invalid argument
```

dlech, 2023-07-26:
I opened #1184 to request a more useful error message. In the meantime, I would suggest printing the default limits/settings to get known valid values and then start tweaking from there.

OverclockedToaster, 2023-07-28:
We already tried that, but it outputs a speed value of 800 which throws the error. If we lower this value to exactly 750, it doesn't throw the error anymore. But if we now try to start the motors at 10% an increase them by 10% every meter, they seem to reach their maximum speed at a speed value of around 200-300. Is there a way to find out the exact speed value required to let the motors run at e.g. 30%?

dlech, 2023-07-28:
You can look up the `rated_max_speed` in https://github.com/pybricks/pybricks-micropython/blob/master/lib/pbio/src/motor/servo_settings.c but the actual max speed will be lower than this and depends on the battery voltage and the load on the motors. For the angular motors the rated max is 1000 deg/s, so a drive base with 43 mm wheels and no gears could not possibly go faster than ${1000 \mathrm{deg} \over \mathrm{s}} \times {1 \mathrm{rot} \over 360 \mathrm{deg}} \times {43 \mathrm{mm} \times \pi \over 1 \mathrm{rot}} \approx 375 \left.\mathrm{mm}\middle/\mathrm{s}\right.$ .

(Related: discussion about how to detect motor overload condition: https://github.com/pybricks/support/issues/1069).

closed 2023-09-24 by OverclockedToaster