pybricks/support #1179: [Bug] Cant start project

ihavenoideawhatimdoingh, 2023-07-25:
**Describe the bug**
[image]
play button cant be clicked

**To reproduce**
i dont know how to reproduce it.
4. See error

**Expected behavior**
i expected it to work since i programmed it on my mac and it worked but now it wont work on windows

**Screenshots**
[image]

laurensvalk, 2023-07-25:
What happens if you disconnect and connect again?

ihavenoideawhatimdoingh, 2023-07-25:
> 

same issue

ihavenoideawhatimdoingh, 2023-07-25:
> What happens if you disconnect and connect again?

do you think it could be because im using chrome?

ihavenoideawhatimdoingh, 2023-07-25:
> 

nope doesnt work

dlech, 2023-07-25:
We need logs in order to troubleshoot the problem. Please see https://github.com/orgs/pybricks/discussions/270 for how to collect such logs.

dlech, 2023-07-25:
Which hub and firmware version are you using?

ihavenoideawhatimdoingh, 2023-07-26:
> Which hub and firmware version are you using?

inventor hub, latest firmware

ihavenoideawhatimdoingh, 2023-07-26:
> We need logs in order to troubleshoot the problem. Please see https://github.com/orgs/pybricks/discussions/270 for how to collect such logs.

couldnt be bluetooth issues since it works on my mac, also it wont work (the logs)

dlech, 2023-07-26:
> latest firmware

So you you are using firmware from https://nightly.link/pybricks/pybricks-micropython/workflows/build/master? What is the exact version?

dlech, 2023-07-26:
> couldnt be bluetooth issues since it works on my mac, also it wont work (the logs)

The Windows and Mac computers have different Bluetooth adapters, so it could be a Bluetooth issue. Without any logs, I'm afraid we can't really provide any further insight.

ihavenoideawhatimdoingh, 2023-07-26:
i cant check logs and im using the newest version

closed 2024-10-02 by laurensvalk