pybricks/support #1186: Fullscreen mode

dmusican, 2023-07-27:
I'm teaching my students to code in pybricks while using a projector in front of a classroom, and I have to keep the font at a reasonable size so that students can see it. The button bar at the top takes a bunch of space that could be used instead for showing a few more lines of code. It would be fabulous if a fullscreen mode existed so that the buttons, sidebar, etc, all get hidden, and so you can get as much space as possible for the code.

The prototype linked in #726 is great.

I suppose the icing on the cake would be some keyboard shortcuts to help manage it, most notably a shortcut to enable/disable fullscreen mode, and another to run the code without having to leave fullscreen mode to do it. Regardless, though, just the prototype linked above is great.

laurensvalk, 2023-07-27:
At the risk of going a bit off-topic:

> I'm teaching my students to code in pybricks

This is awesome, thanks for sharing this! :smile:

dmusican, 2023-07-27:
You're welcome! FYI, on timing: our 3-week summer program ends tomorrow; our last day of coding is today and public demos of projects are tomorrow. So at least in terms of my usage of this feature, I'm not likely to need it next until July 2024, at which point I'll use it every day for 3 weeks. :)