pybricks/support #1189: [Bug] WeDo 2.0 Tilt Sensor (45305) not working with Pybricks v3.3.0b8

fkleon, 2023-07-29:
**Describe the bug**

Running Pybricks v3.3.0b8 on a CityHub with a WeDo 2.0 Tilt Sensor.

* The tilt sensor is detected correctly when running the ["Detecting devices" example](https://docs.pybricks.com/en/latest/iodevices/pupdevice.html#detecting-devices)
* However when instantiating the `TiltSensor` on the correct port, the following error is displayed:

```
OSError: [Errno 19] ENODEV: 

A sensor or motor is not connected to the specified port:
--> Check the cables to each motor and sensor.
--> Check the port settings in your script.
--> Check the line in your script that matches
    the line number given in the 'Traceback' above.
```

Looks like a similar issue to #1131

**To reproduce**
Steps to reproduce the behavior:

1. Plug the tilt sensor into Port B of the City Hub
2. Run the following program:

```python
from pybricks.hubs import CityHub
from pybricks.parameters import Port
from pybricks.pupdevices import TiltSensor

hub = CityHub()

sensor = TiltSensor(Port.B)
print(f"Tilt: {sensor.tilt()}")
```

**Expected behavior**

The tilt sensor to initialise correctly.

**Screenshots**

This is a a CityHub with a `DCMotor` on `Port.A` and a `TiltSensor` on `Port.B`:

```
# hub info
Hub 'crossing' with Pybricks: ('cityhub', '3.3.0b8', 'v1.20.0-23-g6c633a8dd on 2023-07-07') (BLE v2.02.01)
Battery status: 8194 mV / 77 mA

# detect devices
Port.A : ---
Port.B : Wedo 2.0 Tilt Sensor

# motor.settings()
Motor settings: (9000,)

# sensor.tilt()
Traceback (most recent call last):
  File "pap/hub_verify.py", line 16, in <module>
OSError: [Errno 19] ENODEV: 

A sensor or motor is not connected to the specified port:
--> Check the cables to each motor and sensor.
--> Check the port settings in your script.
--> Check the line in your script that matches
    the line number given in the 'Traceback' above.
```

Note the DCMotor is not detected via `PUPDevice` but that's already covered in #1162

laurensvalk, 2023-07-30:
Thanks for reporting! A fix has been pushed which should be available in the next beta release.

fkleon, 2023-07-30:
Thanks for the quick triage and fix @dlech @laurensvalk

I've downloaded the CI build for the branch with the fix (`ci-build-3041-v3.3.0b8-24-g04c7304a`) and can confirm this is now working.

laurensvalk, 2023-07-31:
I hadn't gotten around to trying the fix yet, so I hadn't merged it to master. I suppose I can now. Thanks a lot!

closed 2023-07-31 by laurensvalk