pybricks/support #1195: [Bug] City Hub does not turn itself off after timeout.

laurensvalk, 2023-08-04:
**Describe the bug**
It is supposed to turn off after a few minutes if not connected.

Or maybe it does, but it turns itself back on?

Could be related to the recent ioport changes.

dlech, 2023-08-04:
This was fixed most recently by https://github.com/pybricks/pybricks-micropython/commit/019d95c554706b831ca0af8d654b13dfee94fb56.

It is probably broken again by `pbdrv_ioport_init();` in

```c
void pbdrv_ioport_deinit(void) {
    pbdrv_ioport_init();

    // Turn off power on pin 4 on all ports. This is set to input instead of
    // low to avoid city/move hubs turning back on when button released.
    // as soon as the user releases the power button
    pbdrv_gpio_input(&pbdrv_ioport_pup_platform_data.port_vcc);
}
```

`pbdrv_ioport_deinit()` is called multiple times by the `pbsys_io_ports_poll()` so the VCC keeps getting set to out low instead of input which is what was identified as the issue previously. Also the `pbdrv_init_busy_up/down()` triggered by this could keep the hub from shutting down.

laurensvalk, 2023-10-26:
May be fixed by https://github.com/pybricks/pybricks-micropython/commit/f5d0d81ff5ae1f8794efdf1834b1e25601aea00a.

closed 2023-10-26 by laurensvalk