pybricks/support #1198: Improve pitch measurements by incorporating the gyro

Devilly, 2023-08-05:
### Discussed in https://github.com/orgs/pybricks/discussions/1194

<div type='discussions-op-text'>

<sup>Originally posted by **Devilly** August  3, 2023</sup>
I've got a piece of code that does nothing more than continuously reading the pitch value of a Technic Hub.

```python
from pybricks.hubs import TechnicHub
from pybricks.pupdevices import Motor, UltrasonicSensor
from pybricks.parameters import Button, Color, Direction, Port, Side, Stop
from pybricks.geometry import Axis
from pybricks.robotics import DriveBase
from pybricks.tools import wait, StopWatch

from usys import stdin, stdout

hub = TechnicHub(Axis.X, Axis.Z)
hub.light.on(Color.GREEN)

motor_left = Motor(Port.A, Direction.COUNTERCLOCKWISE)
motor_right = Motor(Port.B)

motors = [
    motor_right,
    motor_left
]

while(True):
    wait(500)

    tilt = hub.imu.tilt()[0]

    print(str(tilt))

    # for motor in motors:
    #     motor.run(1000)
```
An example output of this when putting the hub against a static object is this:

`-47
-47
-47
-47
-47
-47
-47
-47
-47
-47
-47
-47
-47
-47`

This is exactly what I would expect. A constant value. When I then enable the motor lines of code and run again:

`-47
-30
-39
-39
-44
-45
-45
-31
-50
-52
-46
-46
-39`

The values go all over the place. The motors have nothing attached to them so the hub stays in the exact same position as before.

Any idea what's going on or what I can test more to find out the cause?</div>

In short: shakiness of moving motors influences the pitch a lot, which can be improved by taking gyro measurements also into account when calculating the pitch.

laurensvalk, 2023-08-05:
Thanks for opening this issue! We should be able to do this when we include full 3D attitude estimation.

Then we can use its estimated vertical vector instead of the gravity measurement to compute the tilt values.

laurensvalk, 2024-12-03:
This has now been done! See https://github.com/pybricks/support/issues/1907 if you want to try it out early. Otherwise, it will come to the next beta release in a few weeks.