pybricks/support #1203: I have 3 questions

ihavenoideawhatimdoingh, 2023-08-12:
1) how would i connect my hub to my ipad

2) how would i connect my hub to my PC via USB

3) how would i save a file to the hub after 1 or 2?

laurensvalk, 2023-08-13:
iPad isn’t supported because the iOS browsers does not allow Bluetooth connections.

USB isn’t supported on any platform, so I’d recommend using Bluetooth on your PC.

ihavenoideawhatimdoingh, 2023-08-13:
> iPad isn’t supported because the iOS browsers does not allow Bluetooth connections.
> 
> USB isn’t supported on any platform, so I’d recommend using Bluetooth on your PC.

Well how do i upload a file to my hub like in the mindstorms app?

laurensvalk, 2023-08-13:
As mentioned above, USB is not supported in Pybricks.

If you prefer to use USB, consider using the original Mindstorms app.

laurensvalk, 2023-10-23:
Closing as there is no follow-up activity.

closed 2023-10-23 by laurensvalk