pybricks/support #1204: [Question] How to use npm package mpy-cross

emirhalici, 2023-08-14:
**Question**
Hi, I am interested in using mpy-cross. I understand that this is an implementation/port of original mpy-cross tool from MicroPython and built into WASM. Exact package I'm talking about is [mpy-cross-v6](https://www.npmjs.com/package/@pybricks/mpy-cross-v6). However when I install the package with command `npm i @pybricks/mpy-cross-v6` to an empty Javascript directory and try to use it, it fails. I am not sure if I am using it right, and frankly there is not documentation to get help from. 
 
**Context**
I want to be able to cross compile my scripts. Preferably, I'd have implementation specific emit options. My first step is to get it to compile to bytecode. My desired use-case is a web app similar to pybricks for a custom project that involves MicroPython. Is this package not what I'm looking for? If so, can you help me out? I'd be down to opening a PR to add some quick start docs if I can get my copy to work. 

**Reproduce**

package.json
```
{
  "type": "module",
  "dependencies": {
    "@pybricks/mpy-cross-v6": "^2.0.0"
  }
}
```

index.js
```
import { compile as mpyCrossCompileV6 } from '@pybricks/mpy-cross-v6';

const fileName = "hello-world.py";
const fileContents = 'print("Hello world")\n\n';
const options = undefined;
const wasmPath = undefined;

try {
    const result = await mpyCrossCompileV6(fileName, fileContents, options, wasmPath);
    console.log('Compiled, result: ', result);
} catch (e) {
    console.log(e);
}
```

command line result:
```
/Users/emirhalici/Development/WebProjects/web-mpy-cross-test/node_modules/@pybricks/mpy-cross-v6/build/mpy-cross-v6.js:9
var Module=typeof MpyCross!="undefined"?MpyCross:{};var readyPromiseResolve,readyPromiseReject;Module["ready"]=new Promise(function(resolve,reject){readyPromiseResolve=resolve;readyPromiseReject=reject});const args=Module["arguments"];const inputFileName=args[a....
# (Really long unreadable minified Javascript code)
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                             

TypeError: Failed to parse URL from /Users/emirhalici/Development/WebProjects/web-mpy-cross-test/node_modules/@pybricks/mpy-cross-v6/build/mpy-cross-v6.wasm
    at Object.fetch (node:internal/deps/undici/undici:11576:11) {
  [cause]: TypeError [ERR_INVALID_URL]: Invalid URL
      at new NodeError (node:internal/errors:405:5)
      at new URL (node:internal/url:743:13)
      at new Request (node:internal/deps/undici/undici:7132:25)
      at fetch2 (node:internal/deps/undici/undici:10715:25)
      at Object.fetch (node:internal/deps/undici/undici:11574:18)
      at fetch (node:internal/process/pre_execution:239:25)
      at instantiateAsync (/Users/emirhalici/Development/WebProjects/web-mpy-cross-test/node_modules/@pybricks/mpy-cross-v6/build/mpy-cross-v6.js:9:11353)
      at createWasm (/Users/emirhalici/Development/WebProjects/web-mpy-cross-test/node_modules/@pybricks/mpy-cross-v6/build/mpy-cross-v6.js:9:11963)
      at /Users/emirhalici/Development/WebProjects/web-mpy-cross-test/node_modules/@pybricks/mpy-cross-v6/build/mpy-cross-v6.js:9:63499
      at /Users/emirhalici/Development/WebProjects/web-mpy-cross-test/node_modules/@pybricks/mpy-cross-v6/build/index.js:30:13 {
    input: '/Users/emirhalici/Development/WebProjects/web-mpy-cross-test/node_modules/@pybricks/mpy-cross-v6/build/mpy-cross-v6.wasm',
    code: 'ERR_INVALID_URL'
  }
}

Node.js v20.3.0
```
The path input for WASM file is correct. It leads to the actual file. When I specify exact same URL through `wasmPath` same result.

dlech, 2023-08-14:
Here are a few things to consider.

1. You need to make sure your web server is configured to serve `.wasm` files with mime type `application/wasm`. For example, `AddType application/wasm wasm` in `.htaccess` if you are using Apache.
2. When using it on the web, you need to supply the URL of where the `.wasm` file is hosted for the `wasmPath`. For example, we are using webpack, so we use webpack's magic import features to both include the `.wasm` file in the build output and to turn the import into a URL in the code.
3. Also, if using webpack, beware the default `.wasm` handling which tries to import the file as a module, which doesn't work because it is basically an command line executable rather than a javascript module.

emirhalici, 2023-08-15:
Thanks for the reply. I had initially tried to use the package with NodeJS. The logs indicate result of that. 

However, setting the package up on a React app worked (I put the WASM source file on public directory alongside other WASM files, and suplied the URL to it). I've solved my problem now but this question remains unanswered: How does it work on Node? 

By the way, is this package a full port of the original tool? Are code emitters (viper, native, bytecode) & architectures supported out of box?

dlech, 2023-08-17:
I'm not sure why node would be trying to use `fetch()` instead of a file read operation. Maybe node 20 is no longer compatible with the old emscripten we used to create the package? It use to "just work" with node.

And yes, there have been no modifications made to the upstream code.

emirhalici, 2023-08-18:
My goal was to attempt compilation on the web with Viper emitter. It seems to catch ViperTypeError's on compilation indicating Viper is supported. Thanks so much for the amazing work. 

Node issue is likely still needs to be fixed but as my issue was solved, it works on a React project, I'm resolving this issue. 

Thanks again, for the support.

closed 2023-08-18 by emirhalici