pybricks/support #1206: [Question]

Johann007007, 2023-08-15:
**Question**
I try to install Pybricks Firmware on Mindstorms Inventor Hub. 
Error Msg: code.pybricks.com wants to connect to a serial port - no compatible device found.
Can anyone help me?

**Context**
Windows 11, Lego Mindstorms Inventor Hub

laurensvalk, 2023-08-15:
Is your Hub in Update mode, as indicated in the video with the red/blue/green lights?

Can you share a screenshot of the error message?

Johann007007, 2023-08-15:
the light is blinking pink-green-blue
Error Msg:
[image]

dlech, 2023-08-15:
Did you install the appropriate USB driver as described in the instruction? Does the device appear (with the correct driver) in the Windows *Device Manager*?

dlech, 2023-08-15:
Also, if it is looking for a "serial" device instead of a "USB" device, you may be using our old installer instead of https://code.pybricks.com.

Johann007007, 2023-08-15:
I removed the LEGO Technic Large Hub in DFU Mode.
reconnected the hub.
Windows have no driver
Automatisch nach Treibern suchen
not driver found
the same with windows update tool

[image]

[image]

it seems I need a Driver! Where to get it? Any ideas?

Johann007007, 2023-08-15:
Yes, it's all there, but where?
If you install Pybricks with an unprivileged account (which is reasonable), and you want to update the diver with system privileges (which is necessary), Pybrics will not find the driver. It would be helpful to document the filename and directory. 
Many thanks!

laurensvalk, 2023-08-15:
That is good feedback - thanks!

Which steps looked different in your case and how did you resolve it?

dlech, 2023-08-22:
If it isn't there, then it means a driver has already been installed for the DFU device or the hub is not in DFU mode. 

It is only there if there is no driver installed already.

[image]

Once the driver is installed, it moves to *Universal Serial Bus devices*.

[image]

Or if running official LEGO firmware and not in DFU mode, the hub shows up under *Ports*.

[image]

dlech, 2023-08-22:
> There is no directory similar to \Users\pybricks\Documents.

The directory does not matter. It is ignored when you click *Let me pick...*. We could probably delete it or blur it out in the screenshot.

[image]

Johann007007, 2023-08-23:
Thank you very much for the answers. 
The directory was a red herring for me. But now everything works excellently.