pybricks/support #1210: [Question]

fede-ferro, 2023-08-21:
Is there a way to receive accurate data for tilt angle and angular velocity in PrimeHub?

I am a teacher and I am trying to create a robot that uses a PID system to balance on two wheels. I am using the PrimeHub and I have tried to implement the project using both tilt angle and angular velocity. Unfortunately, I am receiving incorrect values from the sensor because the tilt angle is affected by vibrations and provides inaccurate readings, as well as the angular velocity which shows non-zero values even when the robot is stationary. While searching for information, I believe I have understood that this is a bug related to the microcontroller installed in the hub. However, I have not been able to find a solution.

Thank you so much

dlech, 2023-08-23:
You can run this program to get the exact version number.

```python
from pybricks import version

print(version)
```

When I run this program:

```python
...

while True:
    print(*hub.imu.angular_velocity())
    wait(500)
```

I get values like this:

```console
0.1016603 -0.02360439 0.1097347
0.03179574 -0.02327716 -0.03031921
0.03179574 0.04672277 -0.03031921
-0.03833607 0.04668081 -0.03068981
0.03166392 -0.09331918 -0.03068981
-0.108142 0.04646432 -0.03066792
0.101858 0.04646432 0.03933208
-0.108092 -0.09377062 0.03936548
0.03190804 0.1162294 0.1093655
-0.038284 -0.02401483 0.179725
0.03171599 -0.1640148 -0.03027505
0.03173104 -0.02415025 -0.1002823
-0.03826895 -0.02415025 -0.0302823
0.03144702 -0.09416973 -0.03010012
0.03144702 0.1858304 0.1098999
0.1013326 0.1157867 -0.09989763
-0.03866735 0.1157867 -0.02989763
```

which are very close to zero. There is always going to be some amount of noise in the readings. The noise here is just a fraction of a percent of the full scale.

fede-ferro, 2023-08-23:
Thank you very much, I was comparing the obtained result with what I get from the Lego app, so I thought it wasn't working correctly. I will work on the data using code. Thanks!

laurensvalk, 2023-10-23:
The question here seems to be answered, so I'll close this issue. Thanks!

closed 2023-10-23 by laurensvalk