pybricks/support #1214: speaker.play_notes broken in latest beta

MrWonka, 2023-08-28:
In version v3.3.0b8 (Pybricks Beta v2.2.0-beta.8), if you try to play notes as part of an async task, only the first note will play. The first note will instead play for all other notes. 
IE: 
async def addScore():
    await hub.speaker.play_notes(["C4/4", "C5/4"], tempo=140)

laurensvalk, 2023-08-28:
Thanks for reporting! We've recently begun moving some awaitables that consist of multiple awaitables to the Python level (as opposed to implementing everything in C).

See e.g. https://github.com/pybricks/pybricks-micropython/blob/master/bricks/_common/modules/_hub_extra.py.

We should do this for `play_notes` too. We just need an additional hook for the existing note string parsing, or just implement it in Python since performance probably isn't too important here.

I think @dlech already made something in Python back in the ev3dev days. Any thoughts? We could just add it directly to `_hub_extra` as above.

laurensvalk, 2023-10-24:
> The first note will instead play for all other notes.

I hear two distinct notes, so this appears to be working as intended.

I will close the issue for now. Please feel free to re-open if you experience this issue again.

closed 2023-10-24 by laurensvalk