pybricks/support #1215: [Bug] Motor seems to do nothing for a while, then goes ballistic

jaguilar, 2023-08-28:
**Describe the bug**
Motor runs much faster than desired and changes directions without being ordered to.

```
#!/usr/bin/env pybricks-micropython

from pybricks.hubs import EV3Brick
from pybricks.ev3devices import (
    Motor,
)
from pybricks.parameters import Port, Stop, Direction, Button, Color
from pybricks.tools import wait, StopWatch, DataLog
from pybricks.robotics import DriveBase
from pybricks.media.ev3dev import SoundFile, ImageFile
from pybricks.tools import wait, multitask, run_task

arm2 = Motor(Port.C)

async def my_run_target(motor: Motor, speed: int, target: int):
    brick.speaker.beep()
    diff = target - motor.angle()
    mspeed = speed if diff > 0 else -speed

    print(mspeed, diff)
    motor.run(mspeed)

    while abs(motor.angle() - target) > 2:
        await wait(100)

async def test_run_target(motor: Motor):
    await my_run_target(motor, 1, 35)

run_task(test_run_target(arm2))
```

**To reproduce**
Steps to reproduce the behavior:
  1. Follow the directions in #1075 to get nightly pybricks-micropython installed on an EV3 brick. The version I installed is `pybricks-micropython-build-3042-git04c7304a`.
  2. Attach a medium motor to port C of an ev3 brick. An arm is helpful to see the behavior.
  3. Upload and run the program.

**Expected behavior**
Motor would beep, then immediately begin turning at a rate of 1 degree per second clockwise and would run for approximately thirty-five seconds.

**Screenshots**
Video of actual behavior (length: 7s):

https://photos.app.goo.gl/bxHxRv1eL7jcWz238

jaguilar, 2023-08-28:
I have a sneaking suspicion as to why this might be. I wrote this program:

```
async def test_wakeup_rate():
    n = 0
    s = StopWatch()
    s.resume()
    while n < 1000:
        n = n + 1
        await(wait(1))
    s.pause()
    print('Took ', s.time(), ' ms to wake 1000 times')

run_task(test_wakeup_rate())
```

And it printed:

```
Starting: brickrun --directory="/home/robot/ev3test" "/home/robot/ev3test/main.py"
----------
Took  83292  ms to wake 1000 times
----------
Completed successfully.
```

I suspect something in the motor control loop is expecting to wake up more often than this. This seems surprisingly bad, and the problem seems to be with the firmware, not the ev3 device. I wrote a program in full python:

```
>>> import time
>>> def one_thousand_sleep():
...     n = 0
...     while n < 1000:
...             n = n+1
...             time.sleep(0.001)
... 

>>> def time_one_thousand_sleep():
...     before = time.time()
...     one_thousand_sleep()
...     after = time.time()
...     print('Took ', after-before, ' seconds to sleep 1000 times.')
... 
>>> time_one_thousand_sleep()
('Took ', 1.4802501201629639, ' seconds to sleep 1000 times.')
>>> import sys
>>> sys.version
'2.7.13 (default, Sep 26 2018, 18:42:22) \n[GCC 6.3.0 20170516]'
```

laurensvalk, 2023-08-28:
See #1035 for some more details. This is why V3 is not supported on EV3. There aren't currently any plans to do so, I'm afraid.

closed 2023-08-28 by laurensvalk