pybricks/support #1220: [Question] GyroDriveBase straight leading to spinouts

jedierikb, 2023-08-30:
Using GyroDriveBase in beta.pybricks, my robot tries to go straight but spins around in circles instead.

Here's a video:
https://photos.app.goo.gl/YUEu5BcEFzxoxPsu9

I know my `wheel_diameter` and `axle_measurements` aren't that off because straight (& turn too) using DriveBase work great.

Tried changing the front_side parameter to the hub, but that did not change anything.

```
hub = PrimeHub( front_side=Axis.Y ) #exact same misbehavior with Axis.X :-(
left_wheel = Motor( Port.A )
rite_wheel = Motor( Port.B, Direction.COUNTERCLOCKWISE )
db = GyroDriveBase( left_wheel, rite_wheel, wheel_diameter=62.44, axle_track=110 )
db.straight( 500 )
```

Suggestions?

Thank you!

laurensvalk, 2023-08-30:
Can you double check which motor is left and which is right?

If they are flipped, the normal drive base is happy to go straight, but the gyro version would be completely lost.

jedierikb, 2023-08-30:
Thank you, that was the problem.

closed 2023-08-30 by jedierikb