pybricks/support #1229: [Bug] light.blink dosen't work with ColorDistanceSensor

MrWonka, 2023-09-05:
**Describe the bug**
The code "blink" is one of the suggested ones after entering "light", but it throws the error:

AttributeError: 'ColorLight' object has no attribute 'blink'

**To reproduce**
```python
from pybricks.hubs import InventorHub
from pybricks.pupdevices import ColorDistanceSensor
from pybricks.parameters import Port, Color
from pybricks.tools import wait

# Initialize the sensor.
scoreSensor = ColorDistanceSensor(Port.D)
score = 0
hub = InventorHub()

scoreSensor.light.blink(Color.RED, [500,500])
wait(1000)
```

**Expected behavior**
blinky red light

laurensvalk, 2023-09-05:
Thanks for reporting. This is an issue in the auto-completer. The sensor does not have this method.

MrWonka, 2023-09-05:
it totally should

laurensvalk, 2023-09-05:
On the bright side, with async it is now possible to make background light shows yourself :smile:

laurensvalk, 2023-10-23:
Cl

laurensvalk, 2023-10-24:
This was fixed by @dlech in https://github.com/pybricks/pybricks-api/commit/821a3e44555199c4d6b5da6803accef598d7222e and should be coming to a release shortly.

closed 2023-10-24 by laurensvalk