pybricks/support #1230: [Bug] PUPDevice reports Exception [Errno 19] ENODEV on 88005 PoweredUp Lights running v3.3.0b8

BertLindeman, 2023-09-05:
**Describe the bug**
Discover device on version v3.3.0b8 does not see the 88005 Lights and reports ENODEV

**To reproduce**
Steps to reproduce the behavior:
1. Go to beta.pybricks.com
2. Connect a hub at version v3.3.0b8 (or install that here)
3. Connect a 88005 Lights unit (e.g. to Port.A)
4. Run the example code from [detecting-devices](https://docs.pybricks.com/en/latest/iodevices/pupdevice.html#detecting-devices)
5. See the error as the port with the 88005 is reported as ```Port.A : ---```

**Expected behavior**
The same as version v3.3.0b6 does: report the lights
```
Port.A : Powered Up Light
```

**Screenshots**
Non sofar

**Extra info**
The device I used:

1. Inventor-hub at v3.3.0b8
2. Technic-hub at v3.3.0b6

laurensvalk, 2023-09-05:
Could this be a duplicate of https://github.com/pybricks/support/issues/1162?

In other words, does the error only happen with `PUPDevice` or also  with the `Light`?

BertLindeman, 2023-09-05:
Using Light is still OK.
So dup of #1162 seems a good indication.

BertLindeman, 2023-09-09:
Close as Dup of #1162

closed 2023-09-09 by BertLindeman