pybricks/support #1241: [Feature] Support for the full Power Functions IR protocol

kebe67, 2023-09-26:
In pupdevices there is some support for using the Boost Color Distance sensor to send IR signals to the Power Functions IR receiver, using the PFMotors class. But as far as I can see it is limited to only a select few functions. 

Would it be possible to get support for the whole protocol, e.g. similar to how it is done in the Powered Up app, by setting the three first nibbles as hexadecimal digits? There are possibilities in the protocol that I would like to be able to use in Pybricks, like the extended mode.

laurensvalk, 2023-09-27:
You could use the PUPDevice class for that sensor to send an arbitrary 16 bit value. I think this is mode 7 (I'm travelling though, I'd have to double check).

For some hints, try having a look at the source code in Pybricks-micropython/pybricks/pup devices

laurensvalk, 2023-10-23:
This could be a neat example project for the projects page using the `PUPDevice` class as suggested above. We probably couldn't integrate it in the firmware though, so I'll be closing this issue.

closed 2023-10-23 by laurensvalk