pybricks/support #1242: [Question]

NiranjanJ2, 2023-09-28:
**Question**
Write your question here.
Nothing runs even when I am just trying to print a simple phrase.
**Context**
What are you trying to accomplish that led to this question?
I was originally trying to import custom files, but not even the simple print statements register, so I tried just printing, but it still didn't register. I am connected to a charged hub and everything.
**Screenshots**
There is a saying that a picture is worth a 1000 words. Screenshots really help to identify and solve problems.
[image]

laurensvalk, 2023-09-28:
Can you try it with the following link? 

https://beta.pybricks.com

laurensvalk, 2023-10-23:
This can be caused when running the new firmware on the older Pybricks release. Since there is no activity here, I'll close this issue.

closed 2023-10-23 by laurensvalk