pybricks/support #1246: [Technical support]

alessandrofolloni, 2023-10-06:
**Question**
I am encountering the following the error: Traceback (most recent call last):
  File "/Users/alessandrofolloni/path/controllers/main/main.py", line 27, in <module>
    from pybricks.robot import RobotSingleton as _Robot
ModuleNotFoundError: No module named 'pybricks.robot'
WARNING: 'main' controller exited with status: 1.

**Context**
The same code works for a friend, I have obviously checked that pybricks was successfully installed, I work on Python 3.10

import sys
import random
import os
import pathlib
import time
here = pathlib.Path(os.getcwd())
sys.path.append(str(here.parent))
#...
#print(sys.path)
#using pathlib and os and sys modules. easiest way for OS to get cwd.
#pathlib.path gets parent and appends to string current path
#then youre able to import pybricks
# import os
#path = os.getcwd()
#pybricks_directory = path + "\\controllers"
#sys.path.append(pybricks_directory)
sys.path.append("..")  # Add parent directory to sys.path
from pybricks.robot import RobotSingleton as _Robot
# =======================================================================================

from pybricks.hubs import EV3Brick
from pybricks.ev3devices import (Motor, TouchSensor, UltrasonicSensor, GyroSensor)
from pybricks.parameters import Port, Stop, Direction, Button, Color
from pybricks.tools import wait, StopWatch
from pybricks.media.ev3dev import SoundFile, ImageFile

this is the first part of the program with the imports

dlech, 2023-10-06:
Can you share your entire project layout (file and directory structure)?

alessandrofolloni, 2023-10-06:
[image]

The error is present in VsCode but especially in Webots.
I also noticed this problem on robot.py

[image]

dlech, 2023-10-06:
Are you using some kind of EV3 simulator?

alessandrofolloni, 2023-10-06:
On Vscode I have installed the extension Lego Mindstorm EV3 Micropython

dlech, 2023-10-06:
If you are trying to run this on a real EV3, then you should not copy the `pybricks` directory locally. It is built into `pybricks-micropython`. And it would be simpler to put the `robot.py` file in the same directory as `main.py`. Then just `from robot import RobotSingleton` instead of trying to mess with the path.

alessandrofolloni, 2023-10-06:
Is it sufficient to delete it? It's a shared project and it works to the others

closed 2023-10-23 by laurensvalk