pybricks/support #1251: [Bug] Showing specific color no longer works with ColorDistanceSensor

LasseD, 2023-10-09:
**Describe the bug**
The ColorDistanceSensor was able to show the colors red, green and blue. Currently this functionality appears to no longer work. It now always shows green, and only green - no matter which color you choose.

**To reproduce**
Run the following on a Technic Hub. The behaviour also shows on the BoostHub:

    from pybricks.pupdevices import ColorDistanceSensor
    from pybricks.parameters import Button, Color, Direction, Port, Side, Stop
    from pybricks.tools import wait
    start = ColorDistanceSensor(Port.A)
    start.light.on(Color.RED)
    wait(2000)
    
**Expected behavior**
Light should shine red. 

**Shown behavior**
Light shines green.

laurensvalk, 2023-10-09:
If it's showing red, green and blue on the latest version, we should be good, since that's the only colors supported by the sensor :)

laurensvalk, 2023-10-26:
@LasseD,

What happens if you run this snippet?

```python
from pybricks.hubs import ThisHub
from pybricks.iodevices import PUPDevice
from pybricks.parameters import Port, Color
from pybricks.tools import wait

device = PUPDevice(Port.A)
hub = ThisHub()

while True:
    hub.light.on(Color.RED)
    device.write(5, (9,))
    wait(2000)

    hub.light.on(Color.BLUE)
    device.write(5, (3,))
    wait(2000)

    hub.light.on(Color.GREEN)
    device.write(5, (5,))
    wait(2000)
```

The expected result is that the sensor light color matches the hub light.

laurensvalk, 2023-10-26:
We've just updated https://beta.pybricks.com/ to V3.3.0b9. I don't think it should affect this issue, but it's probably worth using the latest if you try the sample above.

legolasse, 2023-10-27:
Hi Laurens.

I tried the code with PUPDevice and it resulted in just green being shown (Technic hub, latest Beta version)

Updating the hub to the latest beta version did unfortunately not change the color that the sensor lights up.

BertLindeman, 2023-10-27:
Can confirm, Laurens. At 3.3.0b9 your snippet shows the same color for the sensor and the hub.

legolasse, 2023-10-27:
That is interesting. Which hub is that?

Shall I perhaps pack out a new sensor and try it on a 51515 hub with the latest beta?

BertLindeman, 2023-10-27:
Did the same with a TecnicHub and the same Color-distance sensor. Neat red-blue-green like the hub led colors.
Previous test was a PrimeHub (I later updated my post above to state is was on a prime hub.)

laurensvalk, 2023-10-28:
Two possibilities:
- Defective sensor? But this is unlikely since you mentioned elsewhere that it does show the RED/BLUE during startup as usual.
- Different sensor version.
- Different sensor firmware version.

Another thing worth trying: Does the BOOST or Powered Up app have a block to set the color? Does it work with that app?

I'll make a debug firmware for you that prints out some information about the sensor.

laurensvalk, 2023-10-28:
- To try it, click this [link](https://github.com/pybricks/pybricks-micropython/actions/runs/6675648445), scroll down, and pick your firmware (Try PrimeHub, TechnicHub, CityHub, or EssentialHub).
- Go to beta.pybricks.com and begin to install the firmware.
- Instead of selecting a hub as usual, choose the file you just downloaded

[image]

- Proceed as usual
- Connect to the hub with Bluetooth.
- Plug in the sensor. This will print out a bunch of information.

laurensvalk, 2023-10-28:
I get this:

```
set baud: 115200
set baud: 2400
type id: 37
num_modes: 11
speed: 115200
fw version: 10000000
hw version: 10000000
new_mode: 10
flags: 00 00 00 00 00 00
mapping: in 10 out 00
mapping: in 10 out 00
Writable: 0
num_values: 8
data_type: 1
new_mode: 9
flags: 00 00 00 00 00 00
mapping: in 10 out 00
mapping: in 10 out 00
Writable: 0
num_values: 2
data_type: 1
new_mode: 8
flags: 00 00 00 00 00 00
mapping: in 00 out 00
mapping: in 00 out 00
Writable: 0
num_values: 4
data_type: 0
new_mode: 7
flags: 00 00 00 00 00 00
mapping: in 00 out 04
mapping: in 00 out 04
Writable: 1
num_values: 1
data_type: 1
new_mode: 6
flags: 00 00 00 00 00 00
mapping: in 10 out 00
mapping: in 10 out 00
Writable: 0
num_values: 3
data_type: 1
new_mode: 5
flags: 00 00 00 00 00 00
mapping: in 00 out 04
mapping: in 00 out 04
Writable: 1
num_values: 1
data_type: 0
new_mode: 4
flags: 20 41 00 00 00 00
mapping: in 10 out 00
mapping: in 10 out 00
Writable: 0
num_values: 1
data_type: 0
new_mode: 3
flags: 00 00 00 00 00 00
mapping: in 10 out 00
mapping: in 10 out 00
Writable: 0
num_values: 1
data_type: 0
new_mode: 2
flags: 00 00 00 00 00 00
mapping: in 08 out 00
mapping: in 08 out 00
Writable: 0
num_values: 1
data_type: 2
new_mode: 1
flags: C8 42 EC 00 00 00
mapping: in 50 out 00
mapping: in 50 out 00
Writable: 0
num_values: 1
data_type: 0
new_mode: 0
flags: 00 00 00 00 00 00
mapping: in c4 out 00
mapping: in c4 out 00
Writable: 0
num_values: 1
data_type: 0
mode combos: 004f
set baud: 115200
```

laurensvalk, 2023-10-28:
```
fw version: 10000000
hw version: 10000000
```

My bet is on different values for this. If they remembered to update it :smile:

laurensvalk, 2023-10-28:
If you have multiple of these sensors, can you verify that they all have the same issue, or just a few?

BertLindeman, 2023-10-28:
My result is identical to yours Laurens.
Did the test on the technic hub with the only one sensor I have of this type.

Bert

legolasse, 2023-11-18:
I can confirm that the code by @laurensvalk works on Spike Prime Hub with latest beta firmware (v3.3.0b9).

However. I have done some digging and found the following: When you start using the distance() function, the light turns from red to green. Is this on purpose, and a side effect of the sensor going into "distance sensing mode"?

dlech, 2023-11-18:
> Is this on purpose, and a side effect of the sensor going into "distance sensing mode"?

Yes, this is intrinsic to the sensor and can't be changed.

laurensvalk, 2023-11-19:
Thanks! Just to make sure:

> I tried the code with PUPDevice and it resulted in just green being shown (Technic hub, latest Beta version)
> 
> Updating the hub to the latest beta version did unfortunately not change the color that the sensor lights up.

So it is working on Prime Hub but still an issue on Technic Hub?

legolasse, 2023-11-19:
I have tested on both Technic Hub and Spike Essential with new and old sensors. All show the same behaviour:
Your code works and shows sensor LED in correct colours.
Calling ".distance()" causes the LED to turn green.

laurensvalk, 2023-11-19:
> Your code works and shows sensor LED in correct colours.

Excellent!

> Calling ".distance()" causes the LED to turn green.

That is right.

So when you said this code turns the light green

```python
from pybricks.pupdevices import ColorDistanceSensor
from pybricks.parameters import Button, Color, Direction, Port, Side, Stop
from pybricks.tools import wait
start = ColorDistanceSensor(Port.A)
start.light.on(Color.RED)
wait(2000)
```

Could there have been some more code in your script that was using the distance function, causing it to turn green?

legolasse, 2023-11-19:
I do not believe so in the pre-beta version I tried initially, as I reduced the code to what I submitted in the first post. However! I might not have restarted the hub prior to running that code, which could have left it in "distance/green" mode when testing the code from the first post.

Shall I try with the non-beta version and see if that recreates the issue, or shall we just be happy that the current beta version works in all applications I have found so far?

laurensvalk, 2024-02-27:
This seems to be resolved. Thanks!

closed 2024-02-27 by laurensvalk