pybricks/support #1268: [Bug] Remote Invocation causes 'Unknown error'

Vinz1911, 2023-10-27:
**Describe the bug**
Calling the Remote function to create a connection to a Remote

**To reproduce**
Steps to reproduce the behavior:
Just call:
```python
remote = Remote()
```

which sometimes throw

```bash
RuntimeError: Unknown error
```

**Expected behavior**
That it does not throw this error

laurensvalk, 2023-10-27:
Does this always happen? Can you share a full script?

The following script seems to work for me:

```python
from pybricks.pupdevices import Remote

remote = Remote()
print("Hello, world!")
```

Vinz1911, 2023-10-27:
No Sorry, this Happens sometimes and totally random

laurensvalk, 2023-10-27:
Would you mind trying if it also happens with this (slightly older) builld?

https://github.com/pybricks/pybricks-micropython/actions/runs/6652294240

If it doesn't happen there, I have a hunch what it might be.

dlech, 2023-11-04:
If my hunch is the same as Laurens' then it should be fixed on the lasted builds too:

https://nightly.link/pybricks/pybricks-micropython/workflows/build/master

laurensvalk, 2024-03-11:
A lot of Bluetooth code has since been released. I'll close it now, but feel free to re-open if this happens again.

closed 2024-03-11 by laurensvalk