pybricks/support #1270: Import commands from Google Colab to Pybricks.

OptimusPrime321, 2023-10-28:
Let's say I have a simple program on Google Colab that gives the output "forward". I also have a custom-built function on the web IDE that takes inputs. Is there any way I can continuously stream the output from Google Colab onto the function in the web IDE of pybricks ? This is what I want, but doing the same for EV3 on VScode will also help. I want the same thing to happen: output from Google Colab to stream onto the function in VScode continuously.

laurensvalk, 2023-10-28:
That sounds related to this. Perhaps for inspiration: https://github.com/orgs/pybricks/discussions/1007

OptimusPrime321, 2023-10-28:
Hi! I think that project is cool! However, I was wondering if you could help me with the question I asked earlier. Also, I was wondering if there's any way to upload PNGs/JPEGs onto the online editor. If you could let me know, that would be great!

laurensvalk, 2023-10-28:
We don't have any ready made examples for that kind of thing, I'm afraid. The link above is probably the closest to what you're asking.

OptimusPrime321, 2023-10-29:
Oh, will have to do something on my own then. Thanks for the help!

closed 2023-10-29 by OptimusPrime321