pybricks/support #1271: Dialogs not visible in Pybricks Beta v2.2.0-beta.9 on Android

dlech, 2023-10-29:
**Describe the bug**
After upgrading to v3.3.0b9 (Pybricks Beta v2.2.0-beta.9) on Android, dialogs no longer appear.

When attempting to open a dialog, the rest of the app goes darker as it should but the dialog is not visible. Taping the dark area returns the app to normal

**To reproduce**
Steps to reproduce the behavior:
1. Update to Pybricks Beta v2.2.0-beta.9 
2. Try to open a dialog like firmware update or about.
3. You can actually scroll the screen to the left and down. The background is all white, but you can eventually find the dialog.

dlech, 2023-11-12:
The fix broke other things and needs to be reverted, so we will need to solve this a different way.

laurensvalk, 2023-11-18:
Traced it back to https://github.com/pybricks/pybricks-code/commit/b219484d31c68350667f653d085464fe105a0ced. This is the first commit where it occured.

laurensvalk, 2023-11-18:
Also confirmed that rolling back to 5.2.1 would work around it, but that would bring back https://github.com/xtermjs/xterm.js/issues/4580.

In any case, hopefully this makes the search for the fix a bit easier.

dlech, 2023-11-18:
Thanks. Tracked down the issue to https://github.com/xtermjs/xterm.js/issues/4805.

closed 2023-11-18 by dlech