pybricks/support #1273: Building a firmware with asyncio included

DrTom, 2023-10-31:
We want to build a firmware with the asyncio module from micropython included. 

To that end we created a symlink from `bricks/_common/modules/uasyncio` to `../../../micropython/extmod/uasyncio`  and adjusted `manifest.py` by adding `include("./modules/uasyncio")`.  We used this approach because the autodetection in `manifest.py` does not work for files in directories and and the directive  `("$(MPY_DIR)/extmod/uasyncio")` from the original `micropython` seems not to work. It says `MPY_DIR` is not set.

When we compile the firmware we can see the following in the log:
```
Including foo as a module.
MPY uasyncio/__init__.py
MPY uasyncio/core.py
MPY uasyncio/event.py
MPY uasyncio/funcs.py
MPY uasyncio/lock.py
MPY uasyncio/stream.py
MPY foo.py
GEN build/frozen_content.c
```
so far so goo. `foo.py` ist just for testing to see if that works. 

Now, when running the firmware using `foo` works:
```
>>> import foo
>>> print(foo.PI)
3.14
```

however, when trying to import (u)asyncio we get an `ImportError`.

```
>>> import asyncio
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'asyncio'
>>> import uasyncio
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'uasyncio'
```

What do we need to do to build `asyncio` (or for that matter any other micropython library) into the firmware?

laurensvalk, 2023-10-31:
Since last week, https://beta.pybricks.com/ includes support for asynchronous multitasking. See `multitask` and `run_task` in the `tools` module.

It does not have all of the `asyncio` module, but it should be enough to do most things. If you think something important is missing, do feel free to let us know.

DrTom, 2023-10-31:
OK, thank you. We will have a look at this. 

I am still curios why our approach including asyncio did not work and what would be necessary to include a larger module based on a directory structure. Do you have any hints?

laurensvalk, 2023-10-31:
As far as I know, `uasyncio` is partially implemented in `C`, not fully as a Python module, so I don't know if this is expected to work.

Modules are included via the manifest file. See this link for more details and possibilities: https://docs.micropython.org/en/latest/reference/manifest.html

More generally, it sounds like you'd like a package with submodules? If you look into this, do let us know what you find :)

laurensvalk, 2023-11-06:
The main question here seems to be answered, so we can close this issue. If specific changes are required for certain build steps like the `Makefile`, feel free to open a dedicated issue.

closed 2023-11-06 by laurensvalk