pybricks/support #1274: Expose raw rgb values

DrTom, 2023-10-31:
We found it valuable to (sometimes) work with raw RGB values instead of HSV/HLS for the following reasons:

* Teaching how a color sensor works internally. 
* Also teaching how to write a color matcher from scratch is much more straight forward. 
* In some cases using raw RGB gives more control and better results. 

Here is a patch https://github.com/DrTom/pybricks-micropython/commit/09b9b161ed3aab4b7865231bc619bd324e452b04 which implements this for the colorsensor provided with the spike sets.

Would you be interested in a pull request or maybe add this on your behalf to pybricks? 

As for a PR: I have no idea how good this implementation is. It works though. I have access to the ColorDistanceSensor and could try to extend it for that one, too.

laurensvalk, 2023-10-31:
Nice! To access raw values, you can also just use the existing `PUPDevice` class.

Since you can use it in Python, you could fairly easily extend the color sensor class that way without building a custom firmware. https://docs.pybricks.com/en/latest/iodevices/pupdevice.html

You can also experiment with all the modes that way to see what they do.

DrTom, 2023-10-31:
Thanks for pointing to the generic PUBDevice. Here is an example how to read the RGB values (maybe it is useful for somebody in the future): 

```
from pybricks.iodevices import PUPDevice
from pybricks.parameters import Port

PBDRV_LEGODEV_MODE_PUP_COLOR_SENSOR__RGB_I = 5  # read 4x int16_t

color_sensor_device = PUPDevice(Port.F)
r, g, b, _ = color_sensor_device.read(PBDRV_LEGODEV_MODE_PUP_COLOR_SENSOR__RGB_I)
print(r, g, b)
```

Much easier than building a custom firmware I guess. Getting the proper mode constant might be the trickiest part; e.g. here https://github.com/pybricks/pybricks-micropython/blob/6d1d9050b23a4bdc5ab426a12965b2f69946b7ae/lib/pbio/include/pbdrv/legodev.h#L316

laurensvalk, 2023-10-31:
For further experimentation, have a look at the ambient HSV mode :)

closed 2023-10-31 by laurensvalk

laurensvalk, 2025-06-19:
It's been a while since I looked at it, but I thought it was just the average of the three or some other kind of redundant information.

zus2, 2025-06-19:
the fourth value in image circles is usually Alpha channel = transparency

laurensvalk, 2025-06-20:
How would the color sensor measure that?

DrTom, 2025-06-22:
Afaik the "I" stands for Intensity and from my observation it is not just redundant information computed in some way from the RGB values. It seems more sensitive and also more likely to "overshoot":by reaching 1024 "earlier" the other. I did not perform any experiments, this is just what I believe to have observed a few times. Since Lego does not share any information about it (and also not the source code of the controller inside) we can not know for sure. 

The hardware of the Sensor suggest that my observation would be reasonable since it features areas with RGB filters but also areas which are not filtered and from which the "I" value would be computed then and should be more sensitive: 

[image]

This image is from an original (quite old) Spike Color Sensor.

DrTom, 2025-06-22:
By the way, the pattern would fit to the TCS34725 Sensor https://ams-osram.com/products/sensor-solutions/ambient-light-color-spectral-proximity-sensors/ams-tcs34725-color-sensor. The "clear" channel ist always above the peak of the RGB ones and more pronounce for the wavelengths where they overlap.

laurensvalk, 2025-06-23:
So it looks like we could potentially improve the reflection mode by using I instead of the average of RGB. If someone wants to dig into this, I would be happy to review or accept a pull request.

```c
// pybricks.pupdevices.ColorSensor.reflection
static mp_obj_t get_reflection(mp_obj_t self_in) {
    // Get reflection from average RGB reflection, which ranges from 0 to 3*1024
    int16_t *data = pb_type_device_get_data(self_in, LEGO_DEVICE_MODE_PUP_COLOR_SENSOR__RGB_I);
    return mp_obj_new_int((data[0] + data[1] + data[2]) * 100 / 3072);
}
static PB_DEFINE_CONST_TYPE_DEVICE_METHOD_OBJ(get_reflection_obj, LEGO_DEVICE_MODE_PUP_COLOR_SENSOR__RGB_I, get_reflection);
```

However, based on what we've seen with other sensors, LEGO device modes don't necessarily always expose the values that are actually useful (this is an understatement), which is why we'd need some data to confirm. For example, this mode already has some kind of reflection compensation (subtracting ambient light), but its measured value is not provided in this mode.

dlech, 2025-06-29:
> is there is any way to take a color measurement with the lights somewhere between full brightness and off?

According to the [LEGO datasheet](https://assets.education.lego.com/v3/assets/blt293eea581807678a/blt62a78c227edef070/5f8801b9a302dc0d859a732b/techspecs_techniccolorsensor.pdf?locale=en-us), no. 

[image]

abliss, 2025-06-29:
@dlech Thanks. I had indeed suspected that there is no _supported_ way. I was more trying to ask if anyone had creative ideas for an unsupported way.

I do need to correct one of my observations above, though. Playing more with DEBUG (mode 8), I can read it for 4 numbers that look similar to RGB_I, except (1) the lights are even brighter in mode 8, and (2) the numbers seem to have a different saturation point. for example, in a well-lit room if I point the sensor out into space, RGB_I returns values that are all 0-2, while DEBUG returns values in the 5-16 range (with more sensitivity). If I point the sensor at a cloudy sky, mode 5 continues to return values around 0-3, but mode 8 returns all 1024. Based on @laurensvalk that "reflection compensation (subtracting ambient light), but its measured value is not provided in this mode", I wonder if mode 8 is actually returning a more raw value before ambient light subtraction? Maybe the actual mechanism of mode 5 is that it is doing a loop like: (a) turn lights off, (b) measure ambient light, (c) turn lights on, (d) measure reflected light, (e) subtract d from b and return it, (f) loop back to (a). This would explain why the lights seem dimmer in mode 5, since they are actually only on a 50% duty cycle.