pybricks/support #1276: [Bug]

Epixx512, 2023-11-02:
**Describe the bug**
There is a red banner saying there was a bluetooth connection error.

**To reproduce**
Steps to reproduce the behavior:
I tried connecting my PrimeHub/InventorHub (my hub is teal). The bluetooth button spun for a while on the website and gave me this error.

**Expected behavior**
I expected my hub to connect succesfully with the computer.
**Screenshots**
There is a saying that a picture is worth a 1000 words. Screenshots really help to identify and solve problems.
[image]

laurensvalk, 2023-11-02:
Did this happen again when you tried it a second time? I see this sometimes but repeating it usually solves it.

Epixx512, 2023-11-15:
> Did this happen again when you tried it a second time? I see this sometimes but repeating it usually solves it.

yes, it did

dlech, 2023-11-15:
What operating system/version are you using? What Bluetooth adapter are you using? 

Also see https://github.com/orgs/pybricks/discussions/270 for more in-depth troubleshooting tips.

dlech, 2023-12-05:
Not all Bluetooth adapters are compatible. See https://github.com/pybricks/support/issues/921.

Also see https://github.com/orgs/pybricks/discussions/270 for more in-depth troubleshooting tips.

ChrisShepherd, 2023-12-05:
It is not a dongle, it is an internal Bluetooth adapter. It's on a brand new computer that I just got Friday.

laurensvalk, 2023-12-14:
Thanks @ChrisShepherd. Since this is a brand new computer, is this perhaps something relevant for you, @JuliaFiMSFT?

Chris, which computer do you have? Can you try the following steps?

--------------

# Chrome Device Log

This can give us more information if there was an operating-system specific problem.

1. Open a new tab or window in your web browser.
2. Type `chrome://device-log` in the address bar and press <kbd>Enter</kbd>.
3. Click **Refresh** if needed.
4. Change the log level to **Debug**.
5. Enable **Detailed Timestamps**
6. Filter for only **Bluetooth** messages.
7. Look for any errors or other useful information.

[image]

Epixx512, 2023-12-15:
ok im no longer having issues

laurensvalk, 2023-12-15:
Someone else has the same issue, so it would be great to keep the issue open. Thanks!

You can still unsubscribe from notifications for this thread if you like.